o-gs/dji-firmware-tools #271: Disabling Hardware pairing in Air 2

RudraSama, 2022-03-16:
I extracted system.img file from Air 2 firmware V01.01.0670_Mavic_Air_2_dji_system.bin.
But I don't how to disable hardware pairing in it.
Please help if someone have any idea about it.

EdHub19, 2023-01-22:
Did you succeed in this?  Have you any instruction on the steps involved?

CedricDLMS, 2024-03-26:
Did you find a way ?

RudraSama, 2024-03-26:
> Did you find a way ?

Yeah

EdHub19, 2024-03-26:
Nope.  Maybe if you unpack software and manually make changes then re-load it you can, but thats way beyond my skill level.  More likely to corrupt software. Regards Edward Terry07736479409

CedricDLMS, 2024-04-09:
Nice , thank for your answer. Was it something hardcore or kind of "attainable" ? I have been checking the sources for a few weeks now...

RudraSama, 2024-04-09:
Well, there are frimwares available for that... you just need to flash that
firmware and calibrate it with your controller(app) and it will fix the
problem. After that you can re-flash the original firmware with dji
assistant.

KTM1551, 2024-04-13:
> Well, there are frimwares available for that... you just need to flash that firmware and calibrate it with your controller(app) and it will fix the problem. After that you can re-flash the original firmware with dji assistant.
> […](#)

Is there a modded FW for this out for the Air 2s? If so whats the FW version? is it available in DDD?

I have DJI SC that is bound to my Air2s but that account is gone to logout, however i want to be able to use the Controller between this Air2s and my other Air2s which has the hacks done to it already, and i can only fly it five times, you konow before i have to use it on the original controller and then i can switch back.. Its really frustrating. I have a super early version of the android fly app, so im good with that. As well as the DJI GO4 app. 

I also have an older version of Litchi that i had a friend put it on for me years ago. I was using my stock fw dji air2s and the park was at the edge of a zone so i couldnt use it and it was all i had, because i had the other one apart alittle to do the front vision sensor board replacement because one of the lenses had a hairline crack on the inside.
So i was playing with Litchi to check out the features and everything, and to my suprise it just was able to do a video mode and it took off to get itself ready i guess, and it worked fine. I had the jeepdoors done so i wasnt worried about that and it was a few yrs back now plus i was 30ft off the ground videoing myself trying out the tracking stuff and video mode like the dji fly app has now.
I do have my iphone with the new version and a mini 3 pro but i stopped updating all my drones 2yrs ago now. I got the avata late and thats how i learned to fly acro actually but it had the update done already and since then ive been looking for info on this stuff lol

RudraSama, 2024-04-13:
Well there is no firmware available to remove this "Bound" status. But if
you want to remove "Calibration Error", you can find firmware for this on
DankDroneDownloader Desktop version.

KTM1551, 2024-05-19:
> Well there is no firmware available to remove this "Bound" status. But if you want to remove "Calibration Error", you can find firmware for this on DankDroneDownloader Desktop version.
> […](#)

Ok so i looked into more, and since the drone with the DH FW on it (Air2s), there is no account connected to it.
However since i bought it used there is a Google Email Account logged in for like the Android side i gues.. I have an iphone so i forget exactly how that works.

Can i remove this Google Account from the controller and use mine or will that mess up something with how the smart controller is the original controller, so it wouldnt matter much if it is Bound, except that i dont have the password.
 
But like i said in the DJI fly app, theres no account logged into it and it works fine

So i was able to un-bind my other Air2s from the DJI Fly App, using it's original N1 controller and my iphone. Is it safe to Bind it to the Smart Controller without knowing that Google email account's password? I dont want to not be able to unbind it in the future, if possible. 

Also so i can understand this more, does the DH's FW remove the dji fly account all together from the process, so that you cant enroll in dji fly care protection, if it was new and eligible? Because i see nothing in the Fly app under Profile. Which im fine with bc i dont want any of that stuff anyway.

KTM1551, 2024-05-19:
> Well there is no firmware available to remove this "Bound" status. But if you want to remove "Calibration Error", you can find firmware for this on DankDroneDownloader Desktop version.
> […](#)

What is the calibration error you're referring too? 

When i replace the front vision sensors, will i need to re-calibrate them or anything like that for them to work properly? Should i buy a clean used part or these new aftermarket? dji parts that look identical. Possibly they are just off new drones, im not sure. Like on ebay. They have the front cameras the 4 together for $50/60. I figure its fine besides that, i might as well fix it. Plus it could interfere with RTH if i forget to change it to off, but then it can also hit something.... so yeah not great lol

Etto110, 2024-10-11:
so did someone find a way to disable hardware pairing? 

i have my magic air 2 bound to  another controller that i don’t have and want to disconnect the  account on it.

Svenny2, 2025-01-20:
I need help disabling hardware pairing on my DJI Mini 3. Is there a patched FW available? Thanks!