o-gs/dji-firmware-tools #276: Reading/repackaging m0900, m1400 and m1401

svarteld, 2022-06-11:
Hello,

Are there possibly tools to convert these three modules into text format with eventual parameter labels, and packaging them back into .bin modules again?

Trying to adjust lightbridge transmitting power settings in WM610 craft and/or GL658B/C remote, but I'm in some deep waters here; long time since I wrote code. Not even sure these are the correct modules.

Latest FW:
WM610_FW_V01.11.01.50_m0900.bin
v01.09.0000_C1_RC_m1400.bin
v01.09.0000_C1_RC_m1401.bin

Any help would be kindly appreciated. /Peter

svarteld, 2022-06-11:
I'll soon have an RF power meter to measure the tinkering, along with some external RF amplifiers, directional patch antennas, etc, so can document what's happening. Got 6x WM610, 4x GL658B, 2x 658C to make some parallel tests with/without amplifiers.

mefistotelis, 2022-06-11:
Get understanding of this script:
https://github.com/o-gs/dji-firmware-tools/blob/master/tests/test_hardcoders_wm610.sh

Then, use the commands separately to extract what you want and change params to what you want.

svarteld, 2022-06-11:
Thanks; this is challenging me :-) Looking at:
WM610_FW_V01.11.01.50_m0900.bin
arm_bin2elf.py
lightbridge_stm32_hardcoder.py

Guess I need something called capstone for lightbridge_stm32_hardcoder.py:

Warning:
This tool requires capstone to disassemble binary bytecode.
Traceback (most recent call last):
  File "./lightbridge_stm32_hardcoder.py", line 142, in <module>
    from amba_sys_hardcoder import eprint, elf_march_to_asm_config, \
  File "./amba_sys_hardcoder.py", line 77, in <module>
    import capstone
ModuleNotFoundError: No module named 'capstone'

mefistotelis, 2022-06-11:
https://github.com/o-gs/dji-firmware-tools/issues/90