o-gs/dji-firmware-tools #280: Wiki addition: WM1601 Mini SE main board photos + specs

samueljmorris, 2022-06-29:
### Table of Contents
[Function](#function)  
[Variants](#variants)  
[Parts](#parts)  
[Programming](#programming)  
[External interfaces](#external-interfaces)  
[Service interfaces](#service-interfaces)  
[Board view](#board-view)  
[Schematics](#schematics)  

# Function

The board contains all processing units of Mavic Mini SE. It contains Flight Controller, but also sensors processing and VPS, plus the main video and FPV pipeline, and radio transceiver.

# Variants

| Marking               | Overview |
| --------------------- | ---------------------- |
| PP002206.02 |  |

# Parts

| Marking               | Amt. | Pkg. | Function | Specification |
| --------------------- | ----:| ---- | -------- | ------------- |
| MIMXRT1064 DVL6B 1N00X ZKGP2120T | 1 | LFBGA196 | i.MX RT1064 Crossover MCU with Arm Cortex-M7 core; used as Flight Controller, also controls gimbal and battery | [description](https://www.nxp.com/part/MIMXRT1064DVL6B) |
| Ambarella AMBARELLA H22-A0-RH A2045 SBHUV-2N5 1N1 H22A34 | 1 | BGA |  Ambarella H22 SoC for consumer applications; integrates image sensor pipeline (ISP), H.265 (HEVC) and H.264 (AVC) encoders, and a Quad core ARM Cortex-A53 CPU; drives image sensor and SD-card, capturing image data and encoding for both SD-card storage and wifi transmission through AR1021X | [description](https://3vpstm1hc6e52739x31131p2-wpengine.netdna-ssl.com/wp-content/uploads/H22-Product-Brief.pdf) |
| winbond W631GU6MB-11 6039DY900012 105WCA TWN | 2 | VFBGA96 | 1G bits DDR3L-1866 SDRAM. Used as RAM for Ambarella H22| [description](https://www.winbond.com/hq/product/specialty-dram/ddr3-sdram/?__locale=en&partNo=W631GU6MB) [datasheet](https://www.winbond.com/resource-files/W631GU6MB_A03.pdf) |
| Atheros AR1021X-CL33D FU0152F9 2015 CHINA | 1 | QFN | IEEE 802.11a/b/g/n 2.4 to 5.8GHz 2T2R WiFi SoC with internal PA and LNA;  | [related spec](https://datasheetspdf.com/pdf-file/1250040/NATER/AR1021X-ITM-UB20-S15PXX0000NV2/1) |
| ACTIVE 8846QM C00BML | 1 |  | Active-Semi Advanced PMIC for Multi-Core Application Processors; supplies stabilized power to all chips on the board | [description](https://www.qorvo.com/products/p/ACT8846) |
| QUALCOMM SMB1350 004 | 1 |  | Programmable switch-mode Li-Ion battery charger and output power controller SMB1350 | [description](https://www.qualcomm.com/products/catalog/quick-charge-30) |
| SN 1708031 TI 8B8 A0HF | 1 | QFN | OPT3101 ToF-based long-range proximity and distance sensor analog front end; drives infrared LED and phototransistor, providing Time of Flight altitude estimation | [description](https://www.ti.com/product/OPT3101) |
| BQ 25700A TI 088 AFT9 | 1 | PWQFN-N32 | BQ25700A, SMBus 1-4 cell NVDC Buck-Boost battery charge controller with system power monitor and processor | [description](https://www.ti.com/product/BQ25700A) [datasheet](https://www.ti.com/lit/ds/symlink/bq25700a.pdf) |

# Programming

| Chips | Firmware | Description |
| --------------------- | ---------------------- | --------------- |
| Ambarella H22 | [[m0100\|Firmware m0100]] | Camera capture and encoding. |
| MIMXRT1064 | [[m0306\|Firmware m0306]] | Flight control. |
| Atheros AR1021X | [[m0905\|Firmware m0905]] | Wi-Fi transmission. |

# External interfaces

| Marking               | Overview |
| --------------------- | ---------------------- |
| TODO |  |

# Service interfaces

The following service pads exist on this board:

| Marking               | Overview |
| --------------------- | ---------------------- |
| TODO  |  |

(service pads are very similar, but not identical, to mini 2)

# Board view

Top of a PP002206.02 board, FCC compliant version:

[image]

[image]

Bottom of the PP002206.02 board:
[image]

[image]

[image]

# Schematics

No schematics available.

ardmed, 2022-12-16:
Ao good explain thank you so match 😍

mefistotelis, 2023-01-09:
ESC board info is still not on wiki, so leaving this open.

Core board info is published:
https://github.com/o-gs/dji-firmware-tools/wiki/WM1605-Main-Processing-Core-Board