o-gs/dji-firmware-tools #288: Thanks brother, but i have same error issue with mavic 2 battery, should i do the same? Do you mean i will use jumper wire from charger terminals to the bms board?

mavicquax, 2022-08-29:
Thanks brother, but i have same error issue with mavic 2 battery, should i do the same? Do you mean i will use jumper wire from charger terminals to the bms board?

Sent from my iPhone

> On Jan 21, 2022, at 3:02 AM, KnusperKnusper ***@***.***> wrote:
> 
> ﻿
> CP2112 and Mavic air 1 instrucion with ERROR 0X00:
> 
> Open Battry, Connect Charger to battery, Bridge PIN +13V to the Side of the battery where +13V would be or charge every cell on its own, connect SDL SCA and GND.
> If error appears switch sda scl.
> 
> Connect in the software, unseal etc...
> 
> —
> Reply to this email directly, view it on GitHub, or unsubscribe.
> Triage notifications on the go with GitHub Mobile for iOS or Android. 
> You are receiving this because you commented.

_Originally posted by @dronythinggoes in https://github.com/o-gs/dji-firmware-tools/issues/196#issuecomment-1018174323_

mavicquax, 2022-08-29:
Hi,
 same problem to me! Have you ever get any idea how to handle the knusperknusper doing. Could it be that there is no power supply for the mavic board? 🙋‍♂️

mefistotelis, 2023-01-09:
Not really sure what that means. I don't think I have proper support of M2 batteries, meaning this is a question about some other battery SW. Closing.

closed 2023-01-09 by mefistotelis