o-gs/dji-firmware-tools #289: Mavic 3 pro

aanestaa, 2022-09-02:
I try to mavic 3 pro gimbal calobration and it not work, some one already try it? Pleace help me.

greg3101, 2022-09-13:
Hello, help please I am in the same situation yet there is the wm260 command for the mavic 3 but no com port or bulk ?

aanestaa, 2022-09-14:
I try to updateing pip and l get error: no backend available, lm a beginner on python and still figue it out

Kaktusoff, 2022-10-23:
Try to enter this commands:
python -m pip install pyusb
*then
pip install libusb
*then download and install this file (libusb-win32-devel-filter-1.2.7.3.exe) and remove all filters
After this maniplations everything should work

Kaktusoff, 2022-10-23:
> Hello, help please I am in the same situation yet there is the wm260 command for the mavic 3 but no com port or bulk ?
Try to do that way
py comm_og_service_tool.py --bulk WM260 GimbalCalib JointCoarse

Kaktusoff, 2022-10-27:
> Hi there I installed everything as you wrote, according to the instructions: python -m pip install pyusb *then pip install libusb *then download and install this file (libusb-win32-devel-filter-1.2.7.3.exe) and remove all filters Then: cd cd dji py setup.py install py comm_og_service_tool.py --bulk WM260 GimbalCalib JointCoarse
> 
> But an error appeared: Error: Unrecognized response to calibration command JointCoarse request.
> 
> How to fix it? [image]

To tell the truth, i don't know. If you will find solution, please write here

Kaktusoff, 2022-10-27:
https://github.com/o-gs/dji-firmware-tools/issues/178

dmytrokulak, 2023-01-29:
I'm having the same issue. What I can see so far from the debugging: instead of getting a response to GimbalCalib request, the drone is sending a request to PC. 

[image]

ptkrpl is 

85, 83, 4, 152, 3, 10, 11, 118, 0, 3, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 27, 0, 253, 255,, 255, 62, 252, 134, 0, 0, 112, 136, 0, 14, 0, 0, 0, 254, 1, 0, 0, 0, 143, 138, 32, 68, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 2, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 245, 86

alives2000, 2023-02-08:
I can help with mavic 3 gimbal calibration, text me on WhatsApp +62 812-8430-9143 or telegram @djianderson

guewald, 2023-02-09:
There is 3 types of error happens when replacing the gimbal:

1- Gimbal IMU data error (40021)
2- Calibration error (40011)
3- Camera not calibrated (1c100204)

Error 1 does not affect any function of the gimbal/camera.
Error 2 is solved when doing specific calibration procedures (some guys are offering this sirvice remotly). 
Error 3 is the only one that i know that afect the function of the camere. With this error The zoom camera does not work so you can only zoom to 4x.

Does anyone know a way to solve the error 3?  Iknow about the DJI softwares but they can be only used with an DJI VPN conection.

lenchik939, 2023-02-09:
Hi!
Dear author please help me calibrate gimbal mavic 3.
COM port is not found.

lenchik939, 2023-02-09:
> При замене подвеса возникает 3 типа ошибок:
> 
> 1- Ошибка данных IMU стабилизатора (40021) 2- Ошибка калибровки (40011) 3- Камера не откалибрована (1c100204)
> 
> Ошибка 1 не влияет ни на какие функции стабилизатора/камеры. Ошибка 2 решается при выполнении определенных процедур калибровки (некоторые ребята предлагают эту услугу удаленно). Ошибка 3 — единственная известная мне ошибка, влияющая на работу камеры. С этой ошибкой не работает зум камеры поэтому можно зумить только до 4х.
> 
> Кто-нибудь знает способ решить ошибку 3? Я знаю о программном обеспечении DJI, но его можно использовать только с подключением DJI VPN.

In my repair experience, if the zoom does not work, then the zoom camera module itself may have been damaged.

guewald, 2023-02-10:
> In my repair experience, if the zoom does not work, then the zoom camera module itself may have been damaged.

No, its not damaged. This function does not work becouse the new camera need to be vinculated with the core board.
If there is someone with DJI VPN acces, i'm interested on it.

nooa, 2023-02-26:
Why is written "**gimbal calobration**" instead of gimbal calibration?!

rsrmdp, 2023-02-27:
> No, its not damaged. This function does not work becouse the new camera need to be vinculated with the core board. If there is someone with DJI VPN acces, i'm interested on it.

Buenas, yo tengo como solucionarlo
+542236513985

SculptorM, 2023-05-27:
> I can help with mavic 3 gimbal calibration, text me on WhatsApp +62 812-8430-9143 or telegram @djianderson

Is there a path you can describe in a reply?   

at this point:  Error: Unrecognized response to calibration command JointCoarse request.

battar348, 2023-12-23:
> حاول هذا التنوع: python -m pip install pyusb *ثم pip install libusb *ثم قم بتنزيل هذا الملف وتثبيته (libusb-win32-devel-filter-1.2.7.3.exe) وقم باختيار جميع الإضافات بعد هذه التلاعبات يجب أن يعمل كل شيء

Did you find a solution to this problem?

battar348, 2023-12-23:
> مرحبًا، لقد قمت بتثبيت كل شيء كما كتبت، وفقًا للتعليمات: python -m pip install pyusb *ثم pip install libusb *ثم قم بتنزيل هذا الملف وتثبيته (libusb-win32-devel-filter-1.2.7.3.exe) إزالة الكل جيلات ثم : cd cd dji py setup.py install py comm_og_service_tool.py --bulk WM260 GimbalCalib JointCoarse
> 
> ولكن ظهر خطأ: خطأ: عضو غير معروف لطلب أمر الاستبدال JointCoarse.
> 
> كيف تصلحها؟ [image]

Did you find a solution to this problem?

elattrib, 2024-05-19:
Hi, any new solution, just replaced my mavic 3 gimbal and camera, getting the error messages": " Camera not calibrated 1C100204 , 1C600204"   and  " Gimbal IMU data error 40021" Thank you

alives2000, 2024-05-20:
> Hi, any new solution, just replaced my mavic 3 gimbal and camera, getting the error messages": " Camera not calibrated 1C100204 , 1C600204" and " Gimbal IMU data error 40021" Thank you
Hi, I don’t think anyone can help with error 1C100204, but I can remove 40021 error
My whatsapp +62 821-7276-9186 or telegram @djianderson

SculptorM, 2024-05-20:
try emailing:   janderssonj17@gmail.com  .  did this for me remote for a fee about a year ago.