o-gs/dji-firmware-tools #291: I have a question about the dji firmware structure

yeohoon1991, 2022-09-16:
I am learning really well through the project wiki. Thank you so much.
However, I have some questions that are not solved.

When a module is extracted from the Phantom 4 Pro (wm331) firmware, 21 files including like wm331_0100_v01.05.15.77_20170531.pro.fw.sig are extracted.

Q1) 
Are all of the module files one common firmware file? In other words, if there is one firmware file that operates general IoT equipment, is there 21 firmware files that make up P4P drones?

Q2)
The chips that make up the p4p drone vary. Examples include Ambara chips, LC1860, and STM32F103. But here the Ambara chip runs on RTOS or Linux and the LC1860 runs on Android 4.4.4. It also states that Flight Controller works with RTOS. What I'm curious about here is that there's one firmware file that runs the drone, but I don't understand that each chip that makes up the drone has a different OS. Is the role independent for each SoC chip regardless of OS? Otherwise, what OS does the firmware file have?

Q3)
I chipped off the flash memory of the P4P drone, but I couldn't get the answer I wanted. As a result, does a drone have several OSs, not one OS?

There were a lot of questions I'm sorry. I talked a lot with my acquaintances to solve this problem, but I couldn't find the answer I wanted. So I asked a question.

Thank you so much for the Wiki Project.

mefistotelis, 2022-09-16:
>  is there 21 firmware files that make up P4P drones?

See wiki. Basics here:
https://github.com/o-gs/dji-firmware-tools/wiki
On board photos, you'll notice there are lost of chips. Most are programmable. And they don't share a common FW.

> is that there's one firmware file that runs the drone

No, there's one FW _package_. A single chip is responsible for extracting the package and distributing to modules around.

> Is the role independent for each SoC chip regardless of OS?

Yes.

> what OS does the firmware file have?

All of them. Programming for each module is based on related chips' SDK.

> does a drone have several OSs, not one OS?

Yes. How would you imagine it having one OS, when there are so any different architectures?
If you buy STM uC, you're not receiving Android or Linux with it, you're getting one Source Dev Kit, and that's it (well maybe 3 in case of STM, but a small number and not really matching any other chips you have).
If you're re-using a chip designated for low-tier mobile phones, you're getting Android SDK with hacks for that chip, and nothing else.

Since you want to make a product now, rather than develop your own SDK which will be better but will take 2 years, you just use what you have.

There are separate teams at DJI working on each chip. They are independent, and that makes their release schedule acceptable for the market. If the teams are independent, they are making the modules in such a way that they work independently as well.

Do you think your PC has a single FW/BIOS? No, it has many binaries. Some are distributed from the larger BIOS binary (like Keyboard Controller FW), some are loaded at startup - ever seen [linux-firmware](https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/) package on your Ubuntu? Not as open-source as you'd imagine, right? Even your graphics card has several firmwares.

yeohoon1991, 2022-09-19:
Thank you so much. Thanks to really help a lot of sense. Finally, I would like to ask you for a perfect understanding.

Q1)
What do you mean by a single chip that acts as "responsible for extracting the package and distributing to modules around"?

Q2)
The independent modules to chip in module that is run by os Even with a different serial communication between each other (majority uart) possible, possible because the same architecture for chips?

Q3)
I checked the 3 in 1 motherboard flash memory chip-off data of the P4P drone.
At this time, performing three role on the main board and system files stored files associated with the boot of that chip is be saved? (Note : https://github.com/o-gs/dji-firmware-tools/wiki/WM330-Receiver-Main-Processor-3-in-1-board).
In other words, each soc, associated with the respective roles of flash memory chips of using a boot data of the chips and files are stored? (For example, the difference between 2GB flash memory on the Ambara Chip and 4GB emmc flash memory on the motherboard)
This is a curiosity that arises from knowing that drones do not operate as a single firmware, but that each chip operates independently.

I'm so sorry and grateful for answering the annoying questions.
I wish you luck on good days.

mefistotelis, 2022-09-19:
> What do you mean by a single chip that acts as "responsible for extracting the package and distributing to modules around"?

Not sure what is unclear. There's a package. Inside are firmwares. Someone has to unwrap the package and act as a postman.

> The independent modules to chip in module that is run by os Even with a different serial communication between each other (majority uart) possible, possible because the same architecture for chips?

What?
If you want others to treat you well, at least put the effort to articulate yourself properly.

The DUML protocol was originally created for UART, but it's a packet-based communication which can be used for high ISO/OSI layers. The DUML packets can be wrapped by TCP/IP, or USB, or CAN. And DJI uses many link layers for the communication. Your mobile device sends and receives DUML by USB.

> each soc, associated with the respective roles of flash memory chips of using a boot data of the chips and files are stored?

Each programmable chip has an associated storage, either internal or external. And that stores the firmware, including boot data. For simple chips, the boot is simple. for complex SoCs, there are multiple levels of boot loaders, and secure boot mechanism which only allows to load signed binary on each level.

How each chips boots, you can find in "Programming reference" issued by chip manufacturer.

Most chips are themselves responsible for updating firmware on their storage. Exceptions are only FPGAs.

yeohoon1991, 2022-09-19:
I felt that I lacked a lot. 
Thank you very much for your help.