o-gs/dji-firmware-tools #292: Signed image file head signature has unexpected size

AntonKukoba1, 2022-09-19:
Mavic Pro 3 firmwares contain 0802 module firmware which is PRAK-UFIE encrypted. But pkghead.signature_size is 384 bytes, i.e. it's RSA-3072. Thus I get:

`Signed image file head signature has unexpected size`

Is there any key which I could try to use to unpack 0802 module?

milenovic, 2023-12-20:
The signature verification issue can be bypassed by not specifying the PRAK key, adding the -f option to force decrypt, and using -k UFIE-2022-08.

ash255, 2024-09-30:
> The signature verification issue can be bypassed by not specifying the PRAK key, adding the -f option to force decrypt, and using -k UFIE-2022-08.

Can you share UFIE-2022-08?
Thank you!

KTM1551, 2024-10-01:
So would this work on the DJI Avata 0802 module firmware file to decrypt it? 

I believe they share a similiar design for that because it was made in the same time period..