o-gs/dji-firmware-tools #296: DJI Air 2s IMU Calibration

umair1988, 2022-09-23:
Hi all,
I took up a project to fix a crashed DJI Air 2s drone. After having replaced the left rear leg and some repairs on the gimbal the aircraft was up again flying but after a few minutes of flight the Air2s displayed an IMU error during the flight and I immediately landed that aircraft safely. I tried the IMU calibration but it always gets stuck at 6% followed by an IMU calibration error. I tried replacing the GPS module, IMU module and GPS flex cable but no joy. The aircraft currently display errors 30055 (IMU not calibrated. Calibrate IMU) and 30056 (IMU calibration incomplete. Recalibrate IMU). I feel it's a software error and some software glitch is stopping me from getting rid of the error. Can the Github community please look into the issue and help me out, please? Also while doing an extensive google search I came across a screenshot of some software kit which hints at calibrating IMU through some software. The screenshot of errors and the potential software that may allow IMU calibration is enclosed. If someone had faced similar errors in the past and or knows something about the software please help out. Many thanks

[image]

[image]

umair1988, 2022-09-23:
Any help please 🙏

rbeck425, 2022-09-28:
These tools on this page (can be downloaded) are what you need. That program above is just a fancy front end to hit buttons. Look on youtube and will find many people using these tools for the Mavic Air 2 with great success. Give it a shot!

rbeck425, 2022-09-28:
https://www.youtube.com/watch?v=6MMpbMhy20Y

umair1988, 2022-09-29:
@rbeck425 thank you for the reply. The video is indeed helpful but it's related to Gimbal calibration. While I have gimbal calibration error but my Air 2s gimbal is pointing in the right direction. My issue is related to IMU calibration which fails no matter what. So question that is there a python code for fixing IMU calibration? It seems that the GUI picture I pasted above has some sort of IMU tool for calibration. So @rbeck425 can you please guide me with regards to IMU calibration tool or python code for same please 🙏? Standing by for your valuable input

rbeck425, 2022-09-29:
Are you saying that the warning message will not go away but the gimbal works correctly after using the python code that is packaged with the DJI-Firmware-Tools?

umair1988, 2022-09-30:
Yes that is what I am trying to say partially. Gimbal is working and warning error 40011 remains for which I am not bothered as it doesn't effect drone operations. My concern is error 30055 & 30056 which are related to IMU module connected to GPS chip. My drone fails to calibrate IMU and error 30055 & 30056 are stopping the drone from taking a flight. Is there any code for calibrating the IMU module?

umair1988, 2022-10-03:
Can someone please guide with IMU calibration error 30055 & 30055 🙏

alxgacc, 2022-10-14:
Who needs remove error 40011 or 40021, you can write me on WhatsApp +62 812-8430-9143

PastorNate, 2023-01-14:
I am with you on this issue and have been searching high and low for this.  Any luck for you?

SandieJacob, 2023-04-03:
To remove 40011 error. Please contact whatsapp +919074728363

Th3SnowyOwl, 2025-01-17:
> [image]
Please tell me where you found this software!

Abiut10, 2026-03-06:
Ninguna solucion aun ?