o-gs/dji-firmware-tools #297: Mavic Mini 2 Gimbal Calibration

rbeck425, 2022-09-27:
What a great set of tools!! I replaced my ESC and core board after my drone fell into a swamp. Everything else seems good including the camera and motors. I am able to fly with GPS, however the camera will not move. It does move around though as normal when booting up. It just sits cocked slightly to the right when it finishes. (I understand this is normal after these type of repairs)

So happy I was able to find this set of tools until I realized that Mavic Mini 2 was not in the list and the commands for the Mini 1 (WM160) were not successful :(

1. Are there any other models that are currently supported in the list that will technically work for the Mavic Mini 2?

2. Is there any truth to rolling back to an earlier version of firmware for the MM2, running the calibration, then upgrading back up to latest firmware? What tools can I use to downgrade?

3. Does anyone have intentions of writing the code to get MM2 working on Gimbal calibration for this set of tools, in the near future?

DJI is scamming people with the inability to repair things ourselves and I could not be more happy to find a supportive community who feels the same way I do. Thank you for ALL who have contributed!

rbeck425, 2022-10-03:
I have done ALOT of digging and this is just the end of the road? You replace a board or gimbal and DJI "has" to do something to pair them? Please someone tell me they know how to take care of such a silly problem.

miniMy22, 2022-11-09:
Hi rbeck425,
I can confirm that the following commands work on mini 2
command 1
py comm_og_service_tool.py --port COM5 WM160 GimbalCalib JointCoarse

command 2
py comm_og_service_tool.py --port COM5 WM160 GimbalCalib LinearHall

providing that your drone is on com5

run dji assistant -consumer series click on log domnload (if I remember well) close assistant, make sure on device manager that your drone is listed as com and replace 5 with whatever com you are going to be on. You will see a gimbal calibration error on DJI fly app but the gimbal works fine.
I fly Litchi and I cant see any error :)

I didn't try rolling back and forth the firmware so I cant comment on that. Here is good video showing the process:
https://www.ifixit.com/Guide/DJI+Mini+2+Gimbal+Calibration/153754
good luck,
J

JJones780, 2023-09-13:
For anyone doing this in Linux,
You have about 5s after /dev/ttyACM0 appears before it goes away ( and switches over to connecting the SD card). I ran each command after reconnecting.
The commands are
sudo python3 comm_og_service_tool.py --port /dev/ttyACM0 WM160 GimbalCalib JointCoarse
sudo python3 comm_og_service_tool.py --port /dev/ttyACM0 WM160 GimbalCalib LinearHall

I get a minor error message about a return value but it worked perfectly. Thanks everyone!

Jordan-Wires, 2024-07-22:
Does anyone have the update firmware files to repair this gimbal?