o-gs/dji-firmware-tools #298: Battery reset pointer appreaciated

nc1914, 2022-10-02:
Hi. I’ve followed the excellent youtube video using a PI to access i2c, and reset the low battery flags CUDEP and CUV, which has resulted in PF=0, resealed my (Mavic Pro) battery - but it still acts like it’s dead. 
I have no leds, no response to the switch or the charger. 
Have I missed a flag please?
It’s charged to 11.5v now, had gone rather low in storage…
Many thanks for your help,
Nathan.

mefistotelis, 2023-01-09:
No idea.
The documentation of BQ30Z55 is available, with that and the ability to read status of the chip it is possible to figure out what was tripped.
Closing.

closed 2023-01-09 by mefistotelis

nc1914, 2023-01-09:
Hi, thanks for your reply, eventually I found out that holding the power button down for 30 seconds reset the bus and the battery is now working normally again.Nathan

cesaralcaide, 2024-01-16:
Hi nc194: "and reset the low battery flags CUDEP and CUV", how did yout did it?, I can't find it in the video