o-gs/dji-firmware-tools #300: Is it possible use this tool for unbind drone frome dji account?

alximiktik, 2022-10-31:


aytacengin, 2023-01-22:
It's like Apple iCloud account. You can not unbind without Device Owner's account. You have to login with Device Owners's (activated account). But as we know, All data is registered on Remote Controller. If you can reset Remote Controller data, can be passed. For your kindly information :)

print-CRUD, 2023-03-01:
Yellow,

I had trouble with dji account at my own drone. Personnel who managed drone, left. Account details lost. IT guys unbinded that drone from account. Now im using that drone in account subscription with different controller (N1) freely. 100~$ solved that trouble. Actually, no new firmware at all. Later i removed other info from MY drone. Like ID, Home point ets.

alximiktik, 2023-03-15:
I know his contact. Thanks

ToniButorac, 2023-03-26:
> > Yellow,
> > I had trouble with dji account at my own drone. Personnel who managed drone, left. Account details lost. IT guys unbinded that drone from account. Now im using that drone in account subscription with different controller (N1) freely. 100~$ solved that trouble. Actually, no new firmware at all. Later i removed other info from MY drone. Like ID, Home point ets.
> 
> Could you give some contacts of you IT guys, who have an $100 to solve this issue?

I have the same problem did you manage to resolve the problem

Michael515613, 2023-04-11:
123

Michael515613, 2023-04-11:
> 

> 我知道他的联系人。+8616603832756 电报
You need to delete the contact information, this is personal privacy

alximiktik, 2023-04-11:
> 
sorry bro. i deleted

Michael515613, 2023-04-11:
> 

OK，Thanks

DateError, 2023-05-06:
hi i can unbind

alximiktik, 2023-05-06:
> hi i can unbind

Thanks. I can do it myself

DateError, 2023-05-06:
how much

alximiktik, 2023-05-06:
> how much

I don't do it remotely, sorry

DateError, 2023-05-06:
give me your contact

alximiktik, 2023-05-06:
software is offline yet :(

DateError, 2023-05-06:
give me your contact plz

weberton78, 2023-05-07:
> o software ainda está offline :(

Tenho interesse como podemos fazer?

Y300Y, 2023-05-18:
> 你好，我可以解绑

请给我你的联系方式

Y300Y, 2023-05-18:
> 黄色的，
> 
> 我在自己的无人机上遇到了 dji 帐户问题。管理无人机的人员离开。帐户详细信息丢失。IT 人员解除了该无人机与帐户的绑定。现在我可以自由地使用不同控制器 (N1) 的帐户订阅中的无人机。100~$解决了那个麻烦。实际上，根本没有新固件。后来我从我的无人机中删除了其他信息。像 ID、Home point 等。

Please give me your contact information

Y300Y, 2023-05-18:
> > 你好，我可以解绑
> 
> 谢谢。我能自己做

Please give me your contact information

weberton78, 2023-05-19:
> > > 你好,我可以解绑
> > 
> > 
> > 谢谢。我能自己做
> 
> Por favor me dê suas informações de contato

rweberton@gmail.com

rsrmdp, 2023-05-19:
https://t.me/rsrmdp

all990, 2023-05-24:
> hi i can unbind

hello, do you can unbind remote? if yes write me skweezyl2@gmail.com

jackson007705, 2023-07-25:
> > hi i can unbind
> 
> Thanks. I can do it myself

hi how to unbind?

jackson007705, 2023-07-25:
> > > 你好，我可以解绑
> > 
> > 
> > 谢谢。我能自己做
> 
> Please give me your contact information

i need help to unbind how to contact u?

DateError, 2023-07-25:
> 

2194439320@qq.com

bananaseason, 2023-07-25:
> > > 你好，我可以解绑
> > 
> > 
> > 谢谢。我能自己做
> 
> Please give me your contact information

bananaseason@gmail.com

Trumble4277, 2023-09-10:
Need help unbinding any one?

araaa11, 2023-09-10:
Hello i need help to unbind 
My Gmail:
ear04240@gmail.com

alximiktik, 2023-09-11:
X

closed 2023-09-11 by alximiktik

Lawesslee, 2023-09-23:
I can help with account locking. You can contact us via aidartoxeiit@gmail.com.

marionunezg, 2023-11-03:
Dear All,

Anyone who can unlink dji fly accounts?

eidmurad, 2023-11-12:
> hi i can unbind

eidmurad@outlook.it

weberton78, 2023-11-13:
Infelizmente não consigo resolver! Estou a procura 😃

Weberton Regino Sobrinho

Y300Y, 2023-12-01:
How is the fee charged?

发自我的iPhone

JrVolt, 2024-09-29:
Someone can recap the situation ? 
Is that possibile ? Is the account login only insider the RC ?!

sekken420, 2024-10-07:
> Someone can recap the situation ? Is that possibile ? Is the account login only insider the RC ?!

Also curious about this. Can anybody unbind the account forcefully?

am93933, 2025-01-30:
Anyone who can unlink dji fly accounts?

Yanison12, 2025-02-12:
> Anyone who can unlink dji fly accounts?

https://yandron.com/products/dji-account-unbind

fabiofrance23, 2025-06-25:
Hello, somone here that can unbind offline a DJI AIR 2S?

avrahamfrz-ux, 2025-10-19:
Need help to unbind avata2 drone without account access.

JamesMalone00, 2025-10-20:
> Need help to unbind avata2 drone without account access.

https://t.me/+Ei53gvGnZLtjNDhl

JamesMalone00, 2026-02-09:
> Hello I have DJI Air 2s Without changing the motherboard, who can remove the account linked to my drone? If you can help me, please contact me via the email below.
> 
> “[brawlprosu486@gmail.com](mailto:brawlprosu486@gmail.com)”

if you have telegram, message them https://t.me/softwaresdji

juansandoval0613, 2026-11-01:
im having thre same issue with my mavic 3, i boughty it second hand but i didint nitice it was still bound to another dvice

LANEWORK, 2026-11-05:
i have mavic 4 pro can anyone please help with unbinding? much appreciated.