o-gs/dji-firmware-tools #303: 40011 error after calibration on Dji Mini 2 and Air 2

alives2000, 2022-11-18:
If someone got error 40011 after calibrating with a script on models dji mini 1, mini 2, air2, air 2S, I found a solution how to remove these error, you can write to me on WhatsApp +62 812-8430-9143 or on my email janderssonj17@gmail.com  I also have a solution how to remove error 40021 after replacing the camera sensor.

roozmoom, 2022-12-04:
Good morning Good morning my friend, how can I update gambal and the error message that appears to me

alximiktik, 2022-12-04:
> 

Give you telegram or Whatsapp

alives2000, 2022-12-05:
@roozmoom write to me on whatsapp +62 812-8430-9143 or telegram @djianderson, I'll help you.

DJI-Solution, 2023-03-22:
For those who have error 40011 or 40021 on your DJI drone after replace gimbal. You will need calibrate firmware that allow you to calibrate and remove error. We also can remotely fix your drone by just connect your drone to your computer and follow our guideline. Telegram me: t.me/djisolution

Ardumen, 2025-01-17:
Напишите какие версии прошивок для каких дронов исправляют эти ошибки, Для dji mavic mini это V01.00.0302.