o-gs/dji-firmware-tools #304: any update MINI 3 PRO?

jayel1919, 2022-11-22:
MINI 3 PRO any update

alives2000, 2022-11-30:
If you need to calibrate the gimbal write to me on WhatsApp +62 812-8430-9143 or telegram @djianderson

mreall, 2022-12-28:
@jayel1919 Were you able to get your mini 3 pro calibrated?

vcollcardenas, 2023-01-10:
Hola!, ALguien logro calibrar el gimbal del Mini 3 Pro

alives2000, 2023-01-10:
@vcollcardenas I can do it, write me on WhatsApp +62 812-8430-9143 or telegram @djianderson

m7rlin, 2023-01-28:
@alives2000 you should stick to Github because your answer can also help other users

mefistotelis, 2023-01-28:
@m7rlin there's a service software which is only available for DJI authorized dealers. This software only works when connected to internal DJI network (ie. over VPN), so it's not really usable for everyone.

Of course, the packets used for calibration can be sniffed or figured out by analyzing the firmware, and then open-source tool could be created. The data downloaded from servers is most likely just a set of calibration parameters adjusted for specific production batch; it is unlikely that these provide a meaningful change to the quality of calibration.

rsrmdp, 2023-01-28:
> @m7rlin there's a service software which is only available for DJI authorized dealers. This software only works when connected to internal DJI network (ie. over VPN), so it's not really usable for [...]

I have a firmware for the air2s that corrects that error.  I would pass it on to you.  since he is one of the few who contributes to the community.  I have used the dji rev tools, but I have no knowledge of programming.  Greetings and tried to contact me through github without success and through mattermost.  Regards, sorry if my English is bad, I speak Spanish

rsrmdp, 2023-01-28:
@mefistotelis I have a firmware for the air2s that corrects that error.  I would pass it on to you.  since he is one of the few who contributes to the community.  I have used the dji rev tools, but I have no knowledge of programming.  Greetings and tried to contact me through github without success and through mattermost.  Regards, sorry if my English is bad, I speak Spanish

pradipwb, 2023-02-17:
hi, can you share the firmaware for removing error.my mail id is pradipmaism@yahoo.co.in

Mrnt, 2023-03-14:
@mefistotelis Do you think that the MPU-6600 has X/Y/Z accell/gyro self test registers like the MPU-6500:
[image] [https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf](url)
It looks like these may be set in the MPU by testing during chip manufacture. If they are read and stored by the drone control board then it would flag a new camera because the different MPU would have different self test values. It would make sense to display the change as an IMU error.

mefistotelis, 2023-03-14:
> Do you think that the MPU-6600 has X/Y/Z accell/gyro self test registers like the MPU-6500:

Similar chips from the same families often share most of the register space.

Whether this is the source for pairing/imu error - don't know.

RudraSama, 2023-04-08:
> @m7rlin there's a service software which is only available for DJI authorized dealers. This software only works when connected to internal DJI network (ie. over VPN), so it's not really usable for [...]

So you are saying this guy @alives2000  is genuine guy?

Jordan-Wires, 2024-07-22:
Does anyone have the update firmware files to repair this gimbal?

madkoo, 2024-09-01:
What is this "special software" and only some random ppl have access to it ? Doesnt seem very legit when they over a paid service over remote and they are have "geniuen" access to dji vpn ?

808statebrazil, 2024-09-06:
> What is this "special software" and only some random ppl have access to it ? Doesnt seem very legit when they over a paid service over remote and they are have "geniuen" access to dji vpn

i need too..