o-gs/dji-firmware-tools #305: Can't detect i2c

Droroid, 2022-11-23:
I attached the phantom 4 pro battery to the 2 and 3 left upper gpio pins and 5th pin to ground same as the pi pinout schmatic. At the pi cmd screen under the dji tools directory i typed i2cdetect -y 1 and recived error "no such file or directory" i tried to type i2cdetect - 2 and the list came up but there is nothing on the "0" section. Only at the "a" section there is a3 detection. Trying to type "dev address axa3 give error nothing came up i cant connect to the battery to unseal and clear the damn PF !

mefistotelis, 2022-11-23:
There are packets exchanged between BMC and uC when the battery is turned on. You can use that to diagnose your connection. There's a tool `piscope` which will help you if you don't have any normal oscilloscope/analyzer.

Also, try writing with sense. If `i2cdetect` binary wouldn't exist, it wouldn't exist regardless of the parameters. And wtf is supposed to be "dev address axa3" ?

For few more clues/hints, try finding a video on YT which I made on "Unlocking SBS-capable battery chip without expensive programmer".

Droroid, 2022-11-23:
> There are packets exchanged between BMC and uC when the battery is turned on. You can use that to diagnose your connection. There's a tool `piscope` which will help you if you don't have any normal [...]

Ive tried to enter "i2cdetect -y 2" insted of 1 and then the list came up. I dont know if thats ok or not. Im not an expert like you. I watched the video 15 times or more. I cant pass the i2cdetect because i cant find the connection (0x0b in your video) if i could get help from someone on telegram or whatsapp or something i can send pictures.

mefistotelis, 2022-11-23:
You seem to have a general issue with I2C connection, with nothing specific to smart batteries. Try searching for posts/articles/videos regarding how I2C works, and how to diagnose issues with it.

Also, how to use command line and bash. Anyone should know that.

For which bus number you should use - I don't know, search Raspberry Pi docs or forums. On my revision of RPI it was 1.

Numbers used by computer tools have certain conventions - ie. `123` is a decimal number, `0123` is an octal one (=83(dec)), and `0x123` is hex (=291(dec)). There's no 'ax' prefix, only '0x'.

Droroid, 2022-11-23:
Ok i will try to learn more about the i2c. I was thinking that the pi has the same GPIO for all of them.. i use pi2 so i searched the gpio pinout and connect the battery board by that pinout. Anyway i think i do something wrong i just cant put my finger on what part of the proccess i do wrong. I am very good with electronics but software its a bit hard for me to catch..

mefistotelis, 2022-11-23:
I think pinouts there are made to be universal, but chips and their connections may differ. But again, see the docs.

GPIO and I2C are separate interfaces inside the SoC, even if connected to the same pin. This also means that if you try to use both at the same time, you may get strange results; for me the I2C was only able to read long packets when I started probing GPIO - that was probably related to impedance at that end of the connection.

Droroid, 2022-11-23:
I followed that sites 
[image]
[image]

Droroid, 2022-11-24:
I manage to solve the i2c problem it was simply the pi seetings (was i2c off i turned it on) now it read the i2c correctly same as your video.. my issue now its the "./comm_sbs_bqctrl.py -vvv --dev_address 0x0b read --help" bring up a screen with no positional arguments. See picture.

Droroid, 2022-11-24:
UPDATE: finaly success!!! Everything well and the battery is good now. My issue was - my knowlageless - i didnt know the basic stuff like update and install PIP smbus on python3 (rpi2) after that everything goes well. Thanks for the help and for this tool !