o-gs/dji-firmware-tools #310: Sleep mode or auto power on/off for autonomous DJI mini SE

kripper, 2022-12-11:
We are looking for a way to 1) power off the Mini SE when battery charging starts and 2) power on when the next mission should be started.

The drone will be charged with a charging pad.

I don't believe implementing a sleep mode is feasible: a low power consumption mode in which the drone can receive a power-on command from the RC while the battery is being charged.

Is there any existing hardware component that could be modified to power on/off the drone?
For instance, it would be handy to auto power off/on the drone when the usb port is connected/disconnected, since our charging pad has only two connectors.
Or should we just add our own hardware to control the power button?
Maybe a IR device powered by an external battery or wired to the led light.
Please share your thoughts.
I prefer to use DJI's existing hardware whenever possible to avoid increasing the weight of the drone.

mefistotelis, 2022-12-13:
> The drone will be charged with a charging pad.

From what I see the drone has an USB port at back:
https://github.com/o-gs/dji-firmware-tools/wiki/WM1605-Main-Processing-Core-Board
So you will already need some kind of hardware to connect there and allow charging.

For turning on - depends what chip is on the other side of USB, and how it's configured.

You can treat most of the DJI drones as an Android phone with motors. In some phones, you can achieve ADB connection while charging - that depends on build configuration. If you're able to enable ADB on the Mini, simple `adb reboot` might work. But personally, I don't even know which chip is on the other side.

If there is DUML connection available when the drone is off, it should be also possible to turn it on. But I doubt there is one.

If I was doing such a project, I'd probably go with a small Arduino - can be configured to do USB or power button depending on needs, and use Power Line Communication to share status with a c&c device, on receive "power on" request.

kripper, 2023-01-21:
Yes. The USB-C port is able to charge, but there is no connection with the drone (and no ADB) when it's powered off.
To power-on the drone, I will "bypass" the power button by soldering two cables to the mainboard.
Now I have another related question:

**DISCLAIMER:** I'm new to electronics and this may be a bad idea/question.

I have a 4-pin USB-C connector (`VBUS`, `GND`, `D+`, `D-`) that is used to quick charge a DJI Mini SE drone using qc2.0.
If feasible and safe, I would like to use the same USB-C cable to also trigger the power on/off button on the drone using a ESP32 with a 8 relay module.
The idea is to bypass the power button by connecting its `COM` and `NO` (?) pins to two pins of the USB-C connector (let's call them `USB1` and `USB2`).
I would then power on the drone by closing the contact between `USB1` and `USB2` for aprox. 6 seconds.
Please note that I would never need to trigger the power button and charge the drone at the same time, but I'm aware that by closing the contact between `COM`, `NO`, `USB1` and `USB2`, and my concern is that this could short circuit and damage the drone's mainboard when triggering the power button or charging the drone.
Is it feasible to reuse two UBS-c pins? If so, which pins may I safely use?

BTW, I'm using a very strong 4-pin 360° magnetic connector and didn't find similar 6-pin versions.
If not feasible/safe, I would need to use a second connector.
I will also need some mechanism to separate the drone from the magnetic connector(s) so it can take off.

mefistotelis, 2023-01-22:
USB - sorry, don't know what that means. USB1 and USB2 is not a good name for pins, as this is how we name two copies of the whole interface.
Regaedless, you're clearly trying to limit your options from the start. Use smallest Arduino, or ESP32 as you mentioned. Let it control the commands instead of analogue control over USB.

Magnets - have you tried using the drone with such magnets attached? Because the drone has a lot if sensors, some of which are sensitive to that, incl but not limited to the compass.

kripper, 2023-01-22:
Edited.

What do you mean by "let it control the commands instead of analogue control over USB"?

The drone and sensores won't be in use while the magnetic connector is connected and I don't think the magnetic connector is much more stronger than the magnets inside the drone motors.

The ESP32 will be

mefistotelis, 2023-01-22:
> The ESP32 is on the base, not on onboard.

Oh, I see.

My idea was to have the enabling uC on the air side, and send it a command like "turn on" - the uC would know that it means to do a proper sequence of shorting MMIO lines (which may be connected to the button).

kripper, 2023-01-22:
Right. it's a matter of weight and battery consumption.
The ideal solution would be to use the existing original hardware with an opensource firmware.