o-gs/dji-firmware-tools #311: P3S Battery repair

Brynod, 2022-12-16:
Hi all,
I am currently working on an over-discharged battery on which i have managed to refresh the cells to the correct voltage; I can communicate with the battery but am unable to Unseal....I'm guessing this is because the default SHA1 key is incorrect; Does anyone happen to know this key, or from where it can be obtained? 

Thanks in advance!

Bryan

Brynod, 2022-12-29:
I FOUND IT!!!

For anyone else working on P3 batteries, the Unseal key is as follows:

0310E6546051541D31584841B05C41A5

Hope this helps!

Bryan

closed 2022-12-29 by Brynod

mefistotelis, 2022-12-29:
Ha, interesting. Looks like that string is plaintext within battery firmware, though with bytes reversed.

this is P3X_FW_V01.11.0010_m1101.bin, see address 0x058E4:

```
0000005810: C2 43 27 17 30 41 B0 12 │ AA D6 B0 12 90 D1 B0 12  ?>'↨0A?↕??>↕??>↕
0000005820: 7A C6 30 41 3F 43 1F 53 │ 7E 4C 0E 93 AB 00 00 00  zu>A?C▼S~L♫??
0000005830: FC 23 0C 4F 30 41 4C 43 │ E2 B2 1D 00 01 24 5C 43  ?#♀O0ALC??↔ ☺$\C
0000005840: 30 41 B0 12 A6 D5 B2 40 │ F4 01 02 17 30 41 B0 12  0A?↕??>@?☺☻↨0A?↕
0000005850: CE D6 30 41 CC 43 01 00 │ 30 41 1E 43 30 40 A0 C8  ??>A~>☺ 0A▲C0@??
0000005860: 03 43 00 13 03 43 00 13 │ 1C 43 30 41 03 43 FF 3F  >C ‼♥C ‼∟C0A♥C??
0000005870: 30 41 70 9A 40 A2 64 32 │ FF 00 00 83 D0 8A 32 32  0Ap?@?d2?  ??>22
0000005880: FF 00 67 0D 03 0D 64 32 │ 01 00 CB 0D 67 0D 32 32  ? g♪♥♪d2☺ _>g♪22
0000005890: 01 00 AB 0A DD 0A 64 32 │ FF 00 79 0A AB 0A 32 32  ☺ ?◙?>d2? y◙?◙22
00000058A0: FF 00 07 1B 01 00 00 00 │ 07 1C 01 00 1E 00 07 19  ? •←☺   •∟☺ ▲ •↓
00000058B0: AC 00 00 00 01 00 00 00 │ 07 1A 01 00 28 00 0D 0A  ?   ☺   •→☺ ( ♪◙
00000058C0: 65 6E 74 65 72 20 69 6E │ 74 6F 20 73 6C 65 65 70  enter into sleep
00000058D0: 0D 0A 00 00 50 48 41 4E │ 54 4F 4D 33 5F 44 4A 49  ♪◙  PHANTOM3_DJI
00000058E0: 42 41 54 00 A5 41 5C B0 │ 41 48 58 31 1D 54 51 60  BAT ?A\?AHX1↔TQ`
00000058F0: 54 E6 10 03 5C 13 5C 13 │ 5A 13 5A 13 5A 13 5A 13  T?►♥\‼\‼Z‼Z‼Z‼Z‼
0000005900: 08 EF CD AB 89 67 45 23 │ 01 47 08 10 32 54 76 98  ◘??>?gE#☺G◘►2Tv?
0000005910: BA DC FE F8 08 10 32 54 │ 76 98 BA DC FE 40 EF CD  ????◘►2Tv????@??
0000005920: AB 89 67 45 23 01 11 22 │ 33 44 55 66 77 00 CC CC  >?gE#☺◄"3DUfw ??
0000005930: 92 00 24 00 AD 00 00 00 │ 12 00 01 02 02 0E 00 14  > $ ?   ↕ ☺☻☻♫ ¶
0000005940: 02 00 02 0F 00 40 9C 02 │ 0B 00 01 02 14 00 B0 04  ☻ ☻☼ @?☻♂ ☺☻¶ ?♦
0000005950: 02 50 00 2E D8 17 02 00 │ 03 05 00 C0 F3 5B 39 BD  ☻P .?>☻ ♥♣ ??[9?
0000005960: D5 86 5D 3C 0F 01 86 A3 │ 4C 6E 43 C4 CC D4 A4 02  ?>]<☼☺??LnC???>☻
0000005970: 14 00 32 30 02 06 58 5F │ 42 41 54 54 45 52 59 02  ¶ 20☻♠X_BATTERY☻
0000005980: 1E 00 64 02 0A 00 02 00 │ 00 00 4A D6 AA D7 86 D6  ▲ d☻◙ ☻   J?>?>?
0000005990: 00 00 1E 00 82 D8 10 11 │ D8 D8 0C 17 FF FF FF FF  > ▲ ??>◄??>↨????
00000059A0: FF FF FF FF FF FF FF FF │ FF FF FF FF FF FF FF FF  ????????????????
00000059B0: FF FF FF FF FF FF FF FF │ 00 02 00 00 30 40 B4 D7  ???????? ☻  0@??
00000059C0: 30 40 46 B4 30 40 2C BD │ 30 40 B4 D7 30 40 B4 D7  >@F?0@,?0@??>@??
00000059D0: 30 40 B0 D7 30 40 7E A8 │ 30 40 78 C5 30 40 B4 D7  >@??>@~?0@xŰ>@??
00000059E0: 30 40 B4 BE 30 40 B4 D7 │ 30 40 B4 D7 30 40 B4 D7  >@??0@??>@??>@??
00000059F0: 30 40 B4 D7 30 40 B4 D7 │ 8C D5                    >@??>@??>?
```

Brynod, 2022-12-30:
Now that IS interesting. I can't help but wonder if the FAS key is stored in this FW also? I haven't managed to find the full access key anywhere unfortunately, but will be happy to help if I can do anything with my spare battery....
Anyone any ideas?

PS great work with the cli @mefistotelis! Thankyou!