o-gs/dji-firmware-tools #316: Mini 3 Pro Gimbal Calibration

gogisoft, 2023-01-01:
Any help getting a replacement gimbal on a Mini 3 Pro working? This repo seems to predate the Mini 3 Pro.

alximiktik, 2023-01-02:
> Any help getting a replacement gimbal on a Mini 3 Pro working? This repo seems to predate the Mini 3 Pro.

@spyman78 telegram

alives2000, 2023-01-04:
@gogisoft you can write me on whatsapp +62-812-8430-9143 or telegram @djianderson, I'll help you

alximiktik, 2023-01-04:
You can buy calibration firmware by 1$ in dank drone downloader

apt-get11, 2023-01-07:
Hello. I can to help you. Gett me your telegram id!

alives2000, 2023-01-07:
@djianderson

Jdm4door, 2023-01-10:
> Hello. I can to help you. Gett me your telegram id!

I have been having trouble making my work after replacing the gimbal arm

[image]

bugalaws, 2023-01-12:
> Hello. I can to help you. Gett me your telegram id!

I have interest in the mavic 3 gimbal calibration too

[image]

bugalaws, 2023-01-12:
[image]
I have interest in the mavic 3 gimbal calibration too

Astenik, 2023-01-21:
Please help dji mini 3 pro Gimball calib Telegramm ID @harh01

aytacengin, 2023-01-22:
Dear friends. Process is very simple. We can calibrate with Special firmwares for calibration version of your device.. You can find some of platforms these firmwares. You have to focus on ARB version. Compatible for your drone ARB ( Anti roll back ) version. 

If you can not find I can manage for you. Write on WhatsApp, http://wa.me/908503468553

m7rlin, 2023-01-28:
you should stick guys to Github because your answer can also help other users

aytacengin, 2023-02-05:
It could be manage with remotly. Very simple. Mavic 3 and Mini3 40011 and 40021 errors can be delete. And can be calibrate..

http://wa.me/908503468553

Ajilforoushan, 2023-02-09:
> 

@rozegar_sakhti

Please add me

greg3101, 2023-02-10:
Whatsapp:
Dji Vpn
+62 812-8430-9143

guewald, 2023-02-10:
If someone have VPN i'm interested +55 51999948379

alives2000, 2023-02-10:
@greg3101 40021 error can be removed only with special software with dji vpn, official dealers can't use dji vpn for remote work, so you don't have dji vpn

greg3101, 2023-02-10:
I just gave you the contact details on wathsap contact him 
Dji Vpn
+62 812-8430-9143

It's hard to understand.

954426470, 2023-02-28:
> 我刚刚给了你关于wathsap的联系方式，联系他Dji VPN +62 812-8430-9143
> 
> 很难理解。
You can send your QR code to my mailbox

akkie2702, 2023-04-26:
> Hello. I can to help you. Gett me your telegram id!

[image]

mrchobbies, 2023-05-17:
Help me with mini3 pro calibration

akkie2702, 2023-05-18:
Hello! I can't calibration my drone! I buy this service

alives2000, 2023-05-19:
@mrchobbies I can calibrate gimbal remotely. Text me on WhatsApp https://wa.me/+6281284309143 or in telegram https://t.me/djianderson

knightRyder18, 2023-08-18:
> Whatsapp: Dji Vpn +62 812-8430-9143

I have just used this number and can verify it's all legit and takes around 10 mins.

$30 is a steal..

954426470, 2023-08-22:
Did you use video?Let me verify

sakis08, 2023-12-24:
Who do you guys recommend for mavic 3 gimbal calibration?

alives2000, 2024-04-01:
> I just gave you the contact details on wathsap contact him Dji Vpn +62 812-8430-9143
> 
> It's hard to understand.

Now my number has changed from +62 812-8430-9143 to +62 857-2762-5367. My telegram account is still djianderson

Jordan-Wires, 2024-07-22:
How can we repair this gimbal and calibrate it ourselves?

davidezotti, 2025-05-09:
I Need calibration and recognize gimbal and camera dji mini 3 thx

Ibastan, 2025-05-22:
> I Need calibration and recognize gimbal and camera dji mini 3 thx

Did it work with alxgacc ?