o-gs/dji-firmware-tools #318: Unable to unseal spark battery with raspberry pi

shileiphys, 2023-01-11:
I'm following [this post](https://ludovic.cool/make-a-custom-battery-for-dji-mavic-pro/) to use Raspberry to fix the dead battery, and I found only `-c BQ40z307` works. However with the command below:
`./comm_sbs_bqctrl.py -v -a 0x0b -c BQ40z307 sealing Unseal`

the security_mode remains __Sealed__

[image]

How to solve the issue? Any other commands to try?

waheedi, 2024-02-04:
Oh man, really :). its already unsealed according to your post, why do you need to re-unseal it? or do you mean you want to seal it, i dont understand your problem.

superdioz, 2024-09-13:
I'm posting here the full command sequence for the dji spark's battery for anyone may need it in the future:
```
./comm_sbs_bqctrl.py --dev_address 0x0b sealing --i32key 0xCCDF7EE0 Unseal
./comm_sbs_bqctrl.py -vvv --dev_address 0x0b trigger ManufacturerAccess.PermanentFailDataReset
./comm_sbs_bqctrl.py --dev_address 0x0b sealing --i32key 0xCCDF7EE0 Seal
```

credits to [dev8edss](https://github.com/dev8edss) for finding [the solution](https://github.com/o-gs/dji-firmware-tools/issues/258#issuecomment-1368695930) for the unsealing.

I also add that if, after a manual charge (just a few volts are sufficient) via pins, the raspberry can't read the address 0x0b with:
`i2cdetect -y 1` you have to connect the battery to a normal charger and click on its button to kick in the led error sequence, after this the rasp detects it and you can do the reset.
I recovered 4 batteries in this way and I can't thank the community enough for it.