o-gs/dji-firmware-tools #325: DJI Air 2S BMS BQ9003 error  code: 110024/30074, Project DOUBLE BATTERY

alanzhch, 2023-02-11:
Does anybody now if the new battery of the air 2s, has a "lock sensor" which recognize if the battery is in the drone or not?
The old dji mavic air batterys had a lock sensor!

Fact: I took the cellpack out of the case and tried to start the drone with only the battey pack inside. The drone starts but takeoff is blocket with the error codes in the titel of that request.

If the project works i will share the documentery how tho build a double battery, out of two original packs.

alanzhch, 2023-02-11:
FYI: I use a CP2112 with DJI Battery Killer, no PF Flag active at this moment.

akhilzid, 2023-02-11:
May be the drone itself triggering the error, air 2 can detect that the battery is properly inserted, just short the terminals near the battery lock to cheat that.

alanzhch, 2023-02-11:
Cheers, i guess you mean the two holes where the metal clips snap in from the battery case, in to the drone case?
If you mean something else with "the terminals" let me know please.

I ordered two new batterycases because i had to cut the originals of, to take the cellpack out.. :) 
Now i try to solder a parallel connection between the two cellpack from the most positive and most negative poles.
The pack which goes in to the drone as before will be charged with the original charger and the upper pack with a charger via xt60 connection. 
The two pack will be connected via xt60 as well and be balanced as before.

alanzhch, 2023-02-11:
The company from singapoore who sells double batterys, dont export the to other nations...
So i have to do it by myself..

akhilzid, 2023-02-11:
This is the "lock sensor" in air 2S shorting this two terminals will detect battery.
[image]

alanzhch, 2023-02-11:
Ahhh one problem solved for the moment. Thank you! 
I will be back next week and will try it out.

Aleksancho1304, 2023-03-30:
what is your current progress?

horstmr, 2024-11-25:
One of the double terminals of the battery lock sensor on the left side has broken in my drone. Do the sensors on both sides have to be shorted so it can take off? 
The drone is currently not with me so I can test this.
Thanks!

alanzhch, 2024-11-25:
Yes, otherwise you will get the "battery installation error" :) Try to solder another pin to it