o-gs/dji-firmware-tools #327: DJI Mini2 Battery : how to clear [TCA] and [TDA] Alarm bits

Scratch-b, 2023-02-13:
Battery not charged  - all leds flashing 3 times per second. I replace battery elements and when use this command:

**$./comm_sbs_bqctrl.py -vvv --dev_address 0x0b -c BQ40z307 -b smbus:1 --short monitor StatusBits**

Opening smbus:1
Importing comm_sbs_chips/BQ40z307.py
Reading simple command at addr=0xb, cmd=0x3, type=uint16, opts={'subcmd': None}
Raw BatteryMode response: 0x6001
BatteryMode: 	0x6001	bitfields	Battery modes and capabilities
[ CapM=0][ ChgM=1][   AM=1][ Rsvd=0][ Rsvd=0][ Rsvd=0][   PB=0][   CC=0] 
[   CF=0][ Rsvd=0][ Rsvd=0][ Rsvd=0][ Rsvd=0][ Rsvd=0][  PBS=0][  ICC=1] 
Reading simple command at addr=0xb, cmd=0x16, type=uint16, opts={'subcmd': None}
Raw BatteryStatus response: 0x48c0
BatteryStatus:	0x48c0	bitfields	Battery's Alarm and Status bit flags
[  OCA=0][  TCA=1][ ResD=0][  OTA=0][  TDA=1][ ResA=0][  RCA=0][  RTA=0] 
[ INIT=1][  DSG=1][   FC=0][   FD=0][                              EC=0] 

I think the problem is [  TCA=1] and [  TDA=1]. No problem to unseal BQ chip, but can't find how to clear this flags.
Any help please

partinis, 2023-06-10:
Hi mefistotelis,
I'm currently facing a rather peculiar issue with one DJI Mini 2 battery. The PF flags are definitely off, the XCHG and XDSG on, and the battery isn't functioning - it blinks three times and stops.
If you have any suggestions or advice regarding my battery issue, I would be grateful. Thank you in advance!
BQStatusBitsMA:
MA.SafetyAlert:         0x00000000      bitfields       Safety Alert bits
[ ResU=0][ ResT=0][ ResS=0][ ResR=0][  UTD=0][  UTC=0][PCHGC=0][ CHGV=0]
[ CHGC=0][   OC=0][ CTOS=0][  CTO=0][ PTOS=0][  PTO=0][ ResH=0][  OTF=0]
[ ResF=0][ CUVC=0][  OTD=0][  OTC=0][ASCDL=0][ ResA=0][ASCCL=0][ Res8=0]
[AOLDL=0][ Res6=0][ OCD2=0][ OCD1=0][ OCC2=0][ OCC1=0][  COV=0][  CUV=0]
MA.SafetyStatus:        0x00000000      bitfields       Safety Status bits
[ ResU=0][ ResT=0][ ResS=0][ ResR=0][  UTD=0][  UTC=0][PCHGC=0][ CHGV=0]
[ CHGC=0][   OC=0][ ResK=0][  CTO=0][ ResI=0][  PTO=0][Res11=0][  OTF=0]
[ ResF=0][ CUVC=0][  OTD=0][  OTC=0][ SCDL=0][  SCD=0][ SCCL=0][  SCC=0]
[ OLDL=0][  OLD=0][ OCD2=0][ OCD1=0][ OCC2=0][ OCC1=0][  COV=0][  CUV=0]
MA.PFAlert:             0x00000000      bitfields       Permanent Fail Alert bits
[  TS4=0][  TS3=0][  TS2=0][  TS1=0][ ResQ=0][ ResP=0][OCECO=0][ ResN=0]
[ ResM=0][ 2LVL=0][ AFEC=0][ AFER=0][ FUSE=0][ ResI=0][DFETF=0][CFETF=0]
[ ResF=0][ ResE=0][ ResD=0][ VIMA=0][ VIMR=0][   CD=0][  IMP=0][   CB=0]
[  QIM=0][ SOTF=0][ Res5=0][  SOT=0][ SOCD=0][ SOCC=0][  SOV=0][  SUV=0]
MA.PFStatus:            0x00000000      bitfields       Permanent Fail Status bits
[  TS4=0][  TS3=0][  TS2=0][  TS1=0][ ResQ=0][  DFW=0][OCECO=0][  IFC=0]
[  PTC=0][ 2LVL=0][ AFEC=0][ AFER=0][ FUSE=0][ ResI=0][DFETF=0][CFETF=0]
[ ResF=0][ ResE=0][ ResD=0][ VIMA=0][ VIMR=0][   CD=0][  IMP=0][   CB=0]
[  QIM=0][ SOTF=0][ Res5=0][  SOT=0][ SOCD=0][ SOCC=0][  SOV=0][  SUV=0]
MA.OperationStatus:     0x00006300      bitfields       Operational Status bits
[ ResU=0][ ResT=0][EMSHT=0][   CB=0][SLPCC=0][SLPAD=0][SLCAL=0][ INIT=0]
[SLEPM=0][   XL=0][CALOO=0][CALOC=0][ACALM=0][ AUTH=0][  LED=0][  SDM=0]
[SLEEP=0][ XCHG=1][ XDSG=1][   PF=0][   SS=0][  SDV=0][           SEC=3]
[ BTPI=0][SMOTH=0][ FUSE=0][ Res4=0][ PCHG=0][  CHG=0][  DSG=0][ PRES=0]
MA.ChargingStatus:      0x0408  bitfields       Charging Status bits
[  VCT=0][ MCHG=0][   SU=0][   IN=0][   HV=0][   MV=1][   LV=0][   PV=0]
[ Res7=0][   OT=0][   HT=0][  STH=0][   RT=1][  STL=0][   LT=0][   UT=0]
MA.GaugingStatus:       0x081455        bitfields       Gauging Status bits
[ ResN=0][ ResM=0][ ResL=0][OCVFR=0][ LDMD=1][   RX=0][ QMax=0][  VDQ=0]
[ NSFM=0][ ResE=0][SLPQMx=0][  QEN=1][  VOK=0][R_DIS=1][ Res9=0][ REST=0]
[   CF=0][  DSG=1][  EDV=0][BAL_EN=1][  TCA=0][   TD=1][   FC=0][   FD=1]
MA.ManufacturingStatus: 0x0028  bitfields       Manufacturing Status bits
[CALTS=0][LT_TS=0][ ResD=0][ ResC=0][ ResB=0][ ResA=0][  LED=0][ FUSE=0]
[  BBR=0][   PF=0][   LF=1][  FET=0][GAUGE=1][  DSG=0][  CHG=0][ PCHG=0]

mefistotelis, 2023-06-10:
Not going to analyze your bit list, sorry.

But if PF is cleared, and the battery still behaves like it has one set - it's likely that the DJI-specific PF2 flag is set. I don't have an implementation of clearing that in my tool, but the "Battery Killer" should be able to clear that.

waheedi, 2024-02-04:
Hello folks,

I hope you are doing well.  And I would like to thank @mefistotelis for this nice tool, I actually happened to have a laptop with this chip used as the battery manager, and have the same problem after successfully unsealing and then fullaccess i have managed to get into the step where i clear my PF, which had only one failure bit which is the CUV one. Now as far as I can see the batteryStatus bits are not writable and you cant change it directly with a command, but rather you would need to flash a new flashdata into the chip with these TDA and TCA set to 0. 

But I'm just guessing so you may try that if you know how or wait until someone else do it :)