o-gs/dji-firmware-tools #328: Actually I am trying to extract Flight Controller module from V01.00.5000 of Phantom 4 pro V2.

RudraSama, 2023-02-18:
After extracting bin file I got multiple files. 2 to 3 files has Linux directories and one file named "wm335_0306xxxx.pro.fw.sig" has flight controller parameters I guess. 
After decrypting it with dji_imah_fwsig.py and 2017 key, it gives output as a bin file. But after analyzing it with binwalk and hex editor I can't really find what this file is and FC param are stored here.
Please help..
(I am new to reverse engineering)

boredafthohaha, 2023-02-22:
Has anybody dumped any of the FHSS stuff from the firmware? I’m looking for SDR related info for demod purposes.

RudraSama, 2023-02-22:
> Has anybody dumped any of the FHSS stuff from the firmware? I’m looking for SDR related info for demod purposes.

I guess there is no support form Phantom 4 pro V2 firmwares.

boredafthohaha, 2023-02-25:
I’ve disassembled a few different firmware versions to get a clue as to how they implemented OFDM and FHSS in previous versions. I guess next step is buying one.

mefistotelis, 2023-02-25:
Extracting WM335 is skipped in CI only because I made a typo in folder name.
If you download firmwares, fix the name and then do:

```
mkdir -p out
pytest tests -rsx -o log_cli=true --log-cli-level=INFO --log-file=out/test-exec.log --junit-xml=out/test-junit.xml --full-scope -k 'wm335-phantom_4_pro_v2-1'
```

It will extract something. Though might need some fixes as due to the typo I ignored that in my local runs as well.