o-gs/dji-firmware-tools #329: Use with Arduino

Pagandy, 2023-02-26:
Hello. I hope you are safe in Ukraine. So sad to see what's happening there. 

Is it possible to use DJI Battery Killer Software with an Arduino Nano in stead of a CP2112? I need to write to the internal battery board to update the data.

Thanks.

mixeysan, 2023-02-26:
Hi mate. Another person is engaged with Arduino. Come to our group - https://t.me/+EJIybgVlnxI4YzM8.
PS This is not my project at Github, please don't write here.

partinis, 2023-06-09:
> Hi mate. Another person is engaged with Arduino. Come to our group - https://t.me/+EJIybgVlnxI4YzM8. PS This is not my project at Github, please don't write here.

Hi mixeysan! Slava Ukraini! I hope this message finds you well. I wanted to reach out and let you know how excited I am about your work on Arduino.
On a separate note, I'm currently facing a rather peculiar issue with one DJI Mini 2 battery. The PF flags are definitely off, the XCHG and XDSG on, and the battery isn't functioning - it blinks three times and stops. I've attempted the Python solution suggested by mefistotelis, but unfortunately, it hasn't yielded any positive results so far.
By the way, I wanted to mention that the Telegram link you provided doesn't seem to be working.
I appreciate your time and expertise. If you have any suggestions or advice regarding my battery issue or if you have an updated link for Telegram, I would be grateful. Thank you in advance!