o-gs/dji-firmware-tools #333: comm_og_service_tool Phantom 4 Calibration

jimmy6616, 2023-03-17:
Ive used this tool on a Phantom 4 Gimbal, the command is accepted but returns a 'Monitoring the progress is only possible from a mobile app, so exiting'

Any help would be good and also just wondering what the JointCourse command actually does

TIA

Jim

mefistotelis, 2023-03-17:
> wondering what the JointCourse command actually does

Then just read the script you're running.

jimmy6616, 2023-03-17:
'Just read the script'? Is it the case I was not waiting for long enough, the 5sec sleep kicks in then wait?
Just looking for help not a programming lesson

jimmy6616, 2023-03-17:
Mefistotelis really trying to get my P4 gimbal calibrated as the camera is out of tilt in pitch and roll. Like I said the command executes then goes back to the ‘C’ prompt after the print command above. Happy for you to Pm me ifit helps at all. I’m not a Python or C programmer just a guy in the UK trying to fix his Phantom and happy to experiment

mefistotelis, 2023-03-17:
As you may read in other "issues" for this project, the auto calibration function does not exist in Ph4.
I don't have anything to say beyond what's already in these other issues.

jimmy6616, 2023-03-17:
I really appreciate the update, I did notice in the script it was focused on the P3, Spark and others but was a bit baffled as the code WM330 exists
It’s a bloody pain that DJI have no reasonable fix for a out of alignment gimbal and I really don’t want to spend another £200 for a damn camera complete when I have it ‘almost’ there
Jim

jose5958, 2024-03-19:
Just make some holes on the chasis to align yaw motor, done.

kramer123, 2024-05-28:
> Ive used this tool on a Phantom 4 Gimbal, the command is accepted but returns a 'Monitoring the progress is only possible from a mobile app, so exiting'
> 
> Any help would be good and also just wondering what the JointCourse command actually does
> 
> TIA
> 
> Jim

On P4, the yaw motor magnet and gimbal ESC are married,  other than that I've changed other motors and it worked fine.   I keep a supply of P4 Gimbals on hand for my customer rebuilds. and also  keeping that Yaw motor or at least the magnet shell in a bag with the ESC board for some rebuilds.   orlando drone repair com

kramer123, 2024-05-28:
> Just make some holes on the chasis to align yaw motor, done.

had to do that once, not the best fix but it did work

KenwoodFox, 2024-08-23:
> > Just make some holes on the chasis to align yaw motor, done.
> 
> had to do that once, not the best fix but it did work

I'm about ready to swap the motor magnet housings, I had the same question about why is WM330 is a listed model if the script cant do anything for it? Regardless. How does the drilling holes work? I just really need the camera to stay straight.. it cant see a thing pointed off all the way to the left.

Surely there's a way to edit the values on the ESC no?