o-gs/dji-firmware-tools #334: Mavic 3 gimbal calibration "Error: No response on calibration command JointCoarse request."

MadddinTribleD, 2023-03-17:
Hey there, 

I'm trying to calibrate the gimbal of a `Mavic 3` (no not `Mini`).

Command i tried:  
`python comm_og_service_tool.py -vv --bulk WM260 GimbalCalib JointCoarse`

Output:  
```
Prepared binary packet:
55 0e 04 66 0a 04 d5 e0 20 02 08 01 21 87
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 38 packets (2185b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 d5 e0 20 02 08 01 21 87
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 12 packets (669b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 d5 e0 20 02 08 01 21 87
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 11 packets (613b), dropped 0 fragments (0b)
No response received.
Error: No response on calibration command JointCoarse request.
```

The problem seems to be that the response i get after `send_request_and_receive_reply()` in `gimbal_calib_request_spark()` has only one byte as payload.  
This results in the check in `comm_mkdupc.py` not responding with an `DJIPayload_Gimbal_CalibRe` object.  
The payload has `1` as data.

Is the Protocol for `Mavic 3` different the once for `Spark`, `Mavic Air` or `Mavic 2 Pro/Zoom`?

If someone could point me do the protocol docs, I'm willing to try filling a PR

mefistotelis, 2023-03-17:
There is no protocol docs. We figure out the protocol (by reverse eng + guessing + experimenting), then we write tools.

The responses were not properly handled earlier (since Mavic Air I think), but the calib was triggered correctly so soone cared.

MadddinTribleD, 2023-03-17:
> There is no protocol docs. We figure out the protocol (by reverse eng + guessing + experimenting), then we write tools.

yea thought so, sad to hear

> The responses were not properly handled earlier (since Mavic Air I think), but the calib was triggered correctly so soone cared.

apparently Mavic 3 uses something else to trigger the calib

alives2000, 2023-03-25:
I can calibrate Mavic 3 gimbal remotely. Text me on WhatsApp [](https://wa.me/+6281284309143) or in telegram [](https://t.me/djianderson)

TymeSlip, 2023-09-07:
I'm having the similar issue with my DJI Mini 2. I just get the prompt of No response on calibration command JointCoarse request. Is there a way to check if this is a hardware issue?

Palvoluss, 2024-05-26:
Hi! I've searched and tried a lot before writing this.

I would like to ask @mefistotelis how we can be sure that the calibration is triggered correctly.

I've run the script multiple times (a lot and in a lot of ways) on both macOS and Ubuntu, with an apparently non-working gimbal of a DJI Mavic Pro 2 and a working Mavic Platinum, but the command 

`./comm_og_service_tool.py --port /dev/ttyACM1 -vv WM220 GimbalCalib LinearHall`

seems to NOT trigger correctly the calibration. 

I've also try to debug the code but i can't figured out anything. There is something we can do to make this work?

mefistotelis, 2024-05-26:
> I would like to ask @mefistotelis how we can be sure that the calibration is triggered correctly.

I haven't looked into that for years at this point. So I don't remember much more that what comes in the code.

Some newer platforms had additional fields in response packets, but I didn't checked that in detail.
Some newer drones also have a bit different format of the request packet, additional fields. But I can't remember any details.

For where the current solution works and where it doesn't - read what other people wrote here in the "issues".

kramer123, 2024-05-28:
> I can calibrate Mavic 3 gimbal remotely. Text me on WhatsApp https://wa.me/+6281284309143 or in telegram https://t.me/djianderson

yes this guy calibrated my Mavic 3,  its not Python software to do that