o-gs/dji-firmware-tools #339: How to lower downgrade firmware DJI MINI 2 from 01.04.000 - 01.06.0200 to 01.03.0000?

Olemarchick, 2023-03-29:
Firmware has expansion .bin, we rename in .tar and we look a suitable self-extractor, (for example FAR) content. We see: the first file of wm161.cfg.sig (about 2 кB) is configuration firmware, 6 other files are the making firmware. That they keep - ordinary people do not know. BUT. One is clear: besides editing of wm161.cfg.sig, namely changes of parameter of recoil of "antirollback="4" on "1" in firmware 01.06.0200 (from 01.06.0200 rolled aside on 01.03.0000) it is needed somehow to change something data in three from six files of firmware (three other absolutely without changes), for example, sewing of "antirollback" related to the number of recoil, and then them somehow, using the instruments of "dji - firmware - tools - master". If all to do right and repacking back in .tar, to rename in .bin, then on an idea it is possible to lower firmware by means of "drone hack" any more fresh to any previous version of firmware, from offered DJI.
One needed: a very good specialist on Pyton and instrument of dji - firmware - tools - master.
UPD1: Yes, and maybe it may need labour for the "signature" of firmware after repacking at DJI. And can be there is some "special key", which does not give it is correct to recooperage firmware, and file of configuration of wm161.cfg.sig
Help please with this question. Can there is a method of lowering of firmware from 01.06.0200 to 01.03.0000. It is very necessary.

rsrmdp, 2023-03-29:
https://github.com/o-gs/dji-firmware-tools/blob/master/README.md

mefistotelis, 2023-03-29:
If you really want to get into such stuff, you need to reach a level where analyzing a code in any language is not a problem for you. And you should definitely be able to use existing tools without anyone holding your hand.

For your idea of lowering antirollback - unfortunately, this is not a fully software feature. Learn about EFUSEs for details.
It can be disabled by software changes, but not the ones you propose.

Olemarchick, 2023-03-30:
Do I paraphrase the question: who can prompt incremental instruction for repacking of files firmware? Id est it is necessary some files (them three) from FW 01.03.0000 it is correct repack in FW 01.06.0200, that it was then possible by means of DH to write in aircraft.

KTM1551, 2023-12-08:
is it actually possible to actually recompile all the files when you're done though, that the aircraft will accept as FW? All i want is to rollback my avata to its original FW because the V2's USB-C port is very sensitive and is getting worn out quickly once i got the Avata after having my FPV for years before that and fpv drone as well. I barely use my Avata because of that but it would be a perfect video drone that can do light Acro for getting Ariel Shots of the Drones or other things vs the an Fpv drone where its good with the O3 but its hard to get it that smooth on a 5" in Acro mode as the Avata can...

softwarelover997, 2025-05-04:
i figured this out, i had a problem with the newer software versions not working correctly on my RC pro, download the firmware zip file, open with winrar or relevant tool, extract .cfg.sig file to be able to edit it, open with notepad, change antirollback value to the same value as your current firmware, you will see there is another identical value which i cannot remember the name of, but it is in the same line,change both. save,paste the edited file in the .zip folder and there you go,now use dronehacks and it will tell you anti rollback values are identical,software upgrade possible