o-gs/dji-firmware-tools #340: Error: Cannot compute image file head signature, auth key 'PRAK' has no private part.

top7chip, 2023-03-31:
Hi
I don't know if what I want to do is possible.
I want to repack a firmware.
I am trying to sign the file "wm160_0000_v01.00.0500_ 20200421.pro.cfg.sig" but it gives me this error "Error: Cannot compute image file head signature, auth key 'PRAK' has no private part."

What am I doing wrong?

C:\Users\top\Desktop\MAVIC FIRM\MINI>dji_imah_fwsig.py -vv -k PRAK-2019-09 -s -i wm160_0000_v01.00.0500_20200421.pro.cfg.sig
wm160_0000_v01.00.0500_20200421.pro.cfg.sig: Opening for creation and signing
wm160_0000_v01.00.0500_20200421.pro.cfg.sig: Warning: Cannot find enc_key ''; scramble key left unencrypted.
wm160_0000_v01.00.0500_20200421.pro.cfg.sig: Packing plaintext chunk '0000'...
OrderedDict([('id', '0000'), ('offset', 0), ('size', 0), ('attrib', 0), ('address', 0), ('reserved', b'\x00\x00\x00\x00\x00\x00\x00\x00')])
wm160_0000_v01.00.0500_20200421.pro.cfg.sig: Decrypted chunks checksum 0x75CED9F3 stored
wm160_0000_v01.00.0500_20200421.pro.cfg.sig: Encrypted data checksum 0x3CD5A0D1 stored
OrderedDict([('magic', b'IM*H'),
            ('header_version', 2),
            ('size', 1888),
            ('reserved', b'\x00\x00\x00\x00'),
            ('header_size', 224),
            ('signature_size', 256),
            ('payload_size', 1408),
            ('target_size', 1888),
            ('os', 0),
            ('arch', 0),
            ('compression', 0),
            ('anti_version', 0),
            ('auth_alg', 2),
            ('auth_key', 'PRAK'),
            ('enc_key', ''),
            ('scram_key', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('name', '0000'),
            ('type', 'CNFG'),
            ('version', 16778496),
            ('date', 538969121),
            ('encr_cksum', 1020633297),
            ('reserved2', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('userdata', b''),
            ('entry', b'\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('plain_cksum', 1976490483),
            ('chunk_num', 1),
            ('payload_digest', b']\x86\xab\x82\xf4\x06\x9eP\xe3\xac\\2q\xe73\xba,w\xe8L\x1apc\x0f\x1b\xa2\xf6\xac_\x8d=\xb2')])
OrderedDict([('id', '0000'), ('offset', 0), ('size', 1378), ('attrib', 0), ('address', 0), ('reserved', b'\x00\x00\x00\x00\x00\x00\x00\x00')])
Error: Cannot compute image file head signature, auth key 'PRAK' has no private part.

top7chip, 2023-03-31:
Thanks for your answer.
Where can I get more information about method 1?

mefistotelis, 2023-03-31:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

davides15, 2026-05-14:
@mefistotelis Let me ask you questions. I'm studying the ndss article about Drone Security, and I get the feeling you participated in the research. There's a point about enabling fastboot mode on the RC N1. It says it's enabled during boot by sending packets. Am I correct in understanding that this refers to packets in the bootrom. It receives packet 60522a7c and responds with 0x44, which is then bootrom accepts the loader. Was a patched loader used to enable fastboot, or does it work differently than on the M3? I'm not asking for a direct answer, just a hint )