o-gs/dji-firmware-tools #343: Mini 3 pro Gimbal Calibration

RudraSama, 2023-04-05:
It is possible to calibrate the gimbal of Mini 3 pro by directly accessing the hardware of drone and triggering it to calibrate it? or just by modifying some values?
I know there are are no markings on core board of drone that how to connect IC directly. And there are no datasheets available on internet about respective ICs. But still anyone have any information which you guys want to share?

alives2000, 2023-04-06:
For mini 3/3pro gimbal calibration uses special software. Same for Mavic 3. I can calibrate gimbal remotely. Text me on WhatsApp https://wa.me/+6281284309143 or in telegram https://t.me/djianderson

atulakbari, 2023-07-10:
mini 3/3pro /Mavic 3 gimbal calibration software.