o-gs/dji-firmware-tools #350: Add aditional infomration to P3 batteries wiki pages

brajomobil, 2023-06-07:
Hi,
Please add correct SDA, SCL,GND pinout for cp2112 and battery killer connection on wiki for P3
Here is pic and more info: https://www.laptopu.ro/community/reset-and-repair-dji-drone-battery/can-detect-dji-phantom-3-battery-chip/

And also this info:
You have to connect power to pcb and balance plug and cp2112 and click unseal + full unseal + clear PF . All in 15 sec.
Or you will have can't connect to SMbus... 

Tnx in advance.
Best regards,
brajomobil

mefistotelis, 2023-06-07:
So you're asking for filling the "service interfaces" section here:
https://github.com/o-gs/dji-firmware-tools/wiki/P3-Battery-Intelligent-board#service-interfaces

..and adding another section about SMBus communication? The SMBus part better matches the FW page (unless HW connections are described):
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m1100#interfaces

You could start by listing all the service pins and providing descriptions to the ones you know; then I will expand that and add to the wiki.

brajomobil, 2023-06-10:
Hi,
Yes. It is i2c protocol for communication. You have to connect 3 wires. GND.SDA.SCL.
I don't know how to add pic to this post. You can take pic from the first link I sent.
After you connect 3 pins to the CP2112 from the battery than you can start program dji battery killer
and you can communicate with BMS chip.

mefistotelis, 2023-06-10:
I don't think you understand me.

I asked for a list of all service pads, like this:
https://github.com/o-gs/dji-firmware-tools/wiki/WM320-Gimbal-top-board#service-interfaces

You create such tables within GitHub using thew following format:

```
| Marking               | Overview |
| --------------------- | ---------------------- |
| 3V3VD | Regulated 3.3V power |
| GND | Ground, connected connected to all other ground pads |
| U0_T | 3.3V UART of Ambarella chip, Transmit line; [[Debug serial interface U0\|Firmware-m0100#debug-serial-interface-u0]] which allows access to Ambarella system console |
| U0_R | 3.3V UART of Ambarella chip, Receive line; |
```

I don't see how the photos you linked to bring anything new - there are already photos of the DJI0617_Rev_04 board which have these pins marked. What is required is a description of the service pads, so that it's unequivocal what the pad markings mean. If I'm mistaken, please point out what exactly is shown there which isn't shown on the wiki.

The wiki is supposed to contain information about each board, but not information about how ti use it with any specific 3rd party software. If the Battery Killer lacks instructions, you should ask for them to be added to that tool.

brajomobil, 2023-06-10:
HI,
Ok than you have to add this it for board: Bottom of the DJI0617_Rev_04 board
On the bottom of board there is i2c communication for BMS chip
| Marking               | Overview |
| --------------------- | ---------------------- |

| GND | Ground, connected connected to all other ground pads |
| SMBD | SDA connection for I2c commutation with BMS
| SMBC | SCL connection for I2c communication with BMS

OK, understand about 3rd party software.

mefistotelis, 2023-06-10:
With some rephrasing, but added:
https://github.com/o-gs/dji-firmware-tools/wiki/P3-Battery-Intelligent-board#service-pads-on-dji0617_rev_04-board

closed 2023-06-10 by mefistotelis

brajomobil, 2023-06-10:
Tnx. 
Please add SDA , and SDC for better understanding.
Best regards,
brajomobil