o-gs/dji-firmware-tools #352: mini 2 working without RC

andreamanera, 2023-06-14:
Good afternoon,
I am Andrea, a student from Italy, Genova, who, together with three other students, is currently developing a project in the field of robotics engineering. In summary the objective of the project is to develop a software to manage UAV fleets autonomously from a ground control station. At this stage of the project we are working with DJI Mini 2 drones and we managed to have more than one drone autonomously flying waypoint missions loaded from PC using the RCs as a bridge between the PC and each drone, in particular to do so we are using Rosetta Drone running on the smartphones connected to the RCs (an open source Github project) and QGroundControl as GCS on the PC. For the next step we would like to transfer what is done by Rosetta on the PC, communicating with the drones using an external antenna connected to the PC, but to do so, keeping the possibility of taking manual control when the RCs are turned on (for safety reasons), we need to have the drones working even without RC, so we thought to edit some firmware parameters, we found the list of the firmware parameters that we can modify for the mini 2, but  there is no documentation about the meaning of the parameters, do you think that it could be possible? If you have any advice we would be grateful. 
Thanks for your help.

mefistotelis, 2023-06-14:
Look at the board:
https://github.com/o-gs/dji-firmware-tools/wiki/WM161-Main-Processing-Core-Board

You will notice antenna connectors in the center part, connected to S1 chip - this is DJI-made SoC called "Sparrow". I contains LTE transmitter, which is reconfigured to create the DJI Ocusync protocol.

How exactly are you planning to re-create the Occusync protocol from a PC? Have your own LTE transmitter and reprogram it to send DJI packets? That's a bit too much work for a student project.

If you don't happen to have a Ocusync transmitter which connects to PC, you may re-consider dropping the RC, and try to just remove the mobile device only - that will likely prove challenging enough.
You will learn a lot looking into the USB protocol, and such a project will be useful to others.

andreamanera, 2023-06-14:
Okay I got you, the plan to remove the RCs was made because of scalability of the system, I mean if the fleet is made of let's say 5 drones it could be uncomfortable to have 5 RCs always connected, I know that also assuming that the problem that I was mentioning is solved we still have the problem of the bandwidth needed to manage all the devices, but we are thinking about solutions for this, anyway I'll keep in mind what suggested by you. I was not specific but the project of which I was talking is a Master thesis project, we are quite familiar with telecommunication and communication protocols even if we are not "expert" of Ocusync yet since we are working on DJI drones from few months, but maybe try to re-create Ocusync on a PC could worth a try (?) Do you know if someone as already tried to do this? 
Anyway thank you a lot for your help, you pointed out things that I was ignoring and gave me new stuff to think about

angusmack, 2023-06-14:
Googling for reverse engineering DJI Ocusync protocol reveals some interesting information, not least this academic paper https://epub.jku.at/obvulihs/download/pdf/6966648?originalFilename=true which starts to do what you want I think plus it's approached with the usual rigor of academia.