o-gs/dji-firmware-tools #353: Phantom 2 Vision Plus CV310 gimbal top board red light issue

fadetonoir, 2023-06-18:
I have a DJI Phantom 2 Vision Plus drone where the camera gimbal shows a steady red light immediately upon power on. According to the manual this is a "Critical error" but I can't find any more information about what can cause it, or anyone who has fixed it before without sending the drone to DJI.

When connecting to the drone via the mobile app there is no data from the drone - no FPV/black screen and no telemetry or other data whatsoever. The drone is otherwise working fine and I have stripped it down to check the cables leading to the gimbal (all are undamaged, and all pins have been tested for continuity with a multimeter), which suggests that the issue is with the [CV310 gimbal top board](https://github.com/o-gs/dji-firmware-tools/wiki/CV310-Gimbal-top-board) (v4 in this case). The board has no apparent physical damage, which leads me to suspect a firmware issue, such as a failed update.

I have tried reflashing firmware using the standard update procedure by putting the file for the latest version on a micro SD card, inserting it into the gimbal and powering on the drone, but this has no effect - the board just powers on to a steady red light on the front, and one flashing slowly on the back (which wasn't visible before disassembly). Also, if I connect the board to a PC with a micro USB cable it is not detected at all.

I have repaired a few of these drones, and a few non-Plus versions as well, when they have had the no-FPV issue caused by a bad firmware update on the live view encoder boards, and even though this is a different issue I was wondering if it would be possible to connect to this board via UART and reflash the firmware. Searching for information on that led me to this repository, but I will admit that even though I've spent some time looking through everything here, most of it is over my head - I do board level repairs and have relatively little knowledge when it comes to firmware. So forgive me if this is the wrong place to ask, but this is the only place I could find information about this board so hopefully someone here could help me fix this, or at least rule out firmware as the cause of the issue.

My specific questions are -

Is anyone familiar with this specific issue on this drone/board and know what the possible cause(s) and fix(es) is/are?
Can anyone tell me how to connect to this board via UART to reflash the firmware or get some sort of boot log from it?
The page for the board (linked above) states "Some schematics were made by enthusiasts; links are TODO" - can anyone send me these schematics?

mefistotelis, 2023-06-18:
> Is anyone familiar with this specific issue on this drone/board and know what the possible cause(s) and fix(es) is/are?

I'm not. But will write based on other Phantoms, which are evolution of this one.

> Can anyone tell me how to connect to this board via UART to reflash the firmware or get some sort of boot log from it?

On the board photos, I clearly see UART0_RX/TX. Tried that?
I'm pretty sure it is the console of the RTOS, like in other drones.

Also, in other drones, it's possible to ask the drone for a log on SD-card, by creating a strange file on it. You may want to read project wiki articles about m0100/m0101 (these module marking were not used until Ph3, but FW has the same base).

Finally, a try at FW update should write a log as well, if only Ambarella is booting. If it doesn't boot, it will not react on the FW at all - no log. If it boots but refuses to update, you should see a log.

> can anyone send me these schematics?

I've seen a try at schematic on rcgroups, but couldn't find it when making the wiki. It's probably under non-obvious name.

Finally, many repair folks are on our mattermost - try asking there. Invite is on one of projects by @Bin4ry.

fadetonoir, 2023-06-19:
Thank you for replying.

> On the board photos, I clearly see UART0_RX/TX. Tried that?

I did - I tried it by soldering just those points to TX and RX on a USB adapter, and then powering the board using the drone and trying to connect via Putty using a baud rate of 115200 (based on how I've been testing the other boards in this drone) but got nothing. If I need to do something else, e.g. also connect 5V from the adapter, or use a different baud rate, then please let me know.

> it's possible to ask the drone for a log on SD-card, by creating a strange file on it.

I read about this on the [M100 page](https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0100) and tried putting a file named ""0x40003000.dji" on the SD card root and powering on the drone, but did not get a log file from it. 

> a try at FW update should write a log as well, 

I haven't seen any log files after firmware update attempts either - I have been checking the SD card because it should rename the update file from firmware_plus.bin to firmware_plus.bin.bak00 when it is completed, but that does not happen and there have been no other files created on the card.

> Finally, many repair folks are on our mattermost - try asking there. Invite is on one of projects by @Bin4ry.

Thanks. I have never used that before but I will take a look.