o-gs/dji-firmware-tools #354: Marvin Pro BQ30z55 FEC Checksum Error

ByteMasterFlinki, 2023-07-17:
I have a Problem, 
If I read the BatteryStatus with
./comm_sbs_bqctrl.py -vvv --bus "i2c:1" -a 0x40 -c BQ30z55 read BatteryStatus
I get an error
Opening i2c:1
Importing comm_sbs_chips/BQ30z554.py
Reading simple command at addr=0x40, cmd=0x16, type=uint16, opts={'subcmd': None}
Raw BatteryStatus response: 00 00 00
Retrying due to error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum
Raw BatteryStatus response: b3 a2 f2
Retrying due to error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum
Raw BatteryStatus response: a3 00 b4
Error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum
Is my command not right what can I try

Thanks

ps: if my post is not correct, sorry it is my first time using GitHub

mefistotelis, 2023-07-17:
0x40 is the uC. `bqctl` script is designed to talk to BMS chip, not the accompanying uC.

mefistotelis, 2023-07-17:
BMS typically is able to work at low voltage.

Read the chip reference - maybe it's in shutdown mode. Or maybe needs reboot.