o-gs/dji-firmware-tools #367: Mavic 3T Gimbal calibration is not supported

linkuha, 2023-10-25:
Mavic 3 (WM260) product code is didn't fit.
Are there any ideas on how to flash Mavic 3T or to calibrate his gimbal using the OGs Service Tool (GimbalCalib)?

coptersafe, 2023-10-25:
врядли тебе здесь подскажут)

alxgacc, 2023-10-26:
I can calibrate the gimbal on this model remotely. Write to me on WhatsApp https://wa.me/+6281284309143 or telegram https://t.me/djianderson

alives2000, 2023-10-26:
I can calibrate the gimbal on this model remotely. Write to me on WhatsApp https://wa.me/+6281284309143 or telegram https://t.me/djianderson

sakis08, 2023-12-24:
Have you found any solution yet?

abdulrahmankrij, 2024-08-15:
> لم يكن رمز منتج Mavic 3 (WM260) مناسبًا. هل هناك أي أفكار حول كيفية تثبيت Mavic 3T أو معايرة gimbal باستخدام أداة OGs Service Tool (GimbalCalib)؟

If you were able to solve the problem, tell us.