o-gs/dji-firmware-tools #370: Website of LANRUISI connectors manufacturer

BohdanHolyshevskyi, 2023-11-13:
Some connectors are labeled as "LANRUISI" but I can't find its website. Also google doesn't return any meaningful results for "LANRUISI".

mefistotelis, 2023-11-13:
Nothing unusual for Chinese stuff. Just search the thing you want on Aliexpress.

yangolan, 2024-08-18:
@BohdanHolyshevskyi have you had any luck finding these connectors? I cannot find a replacement part for the 0.4mm 50 pin FPC connector on dji mavic

BohdanHolyshevskyi, 2024-08-18:
@yangolan try to search at https://www.digikey.com or https://www.lcsc.com. I was able to find some of the connectors. Just filter by pitch and pin count.

Ardumen, 2025-01-17:
На дронах серии mavic mini 1/2 используют FPC разъемы AXE524127, AXE534127, AXE516127 (гибкая плата 3 в 1). На 50 пинов она будет маркироваться AXE550127.