o-gs/dji-firmware-tools #371: DJI battery killer

flf3, 2023-11-28:
howdy.  i have a couple batteries that will not work.  I have the board and jumpers correctly attached to the battery.  I can not unseal.  any advice?  thanks,  Fil 

[image]

EagleeyeKai, 2023-12-21:
Hi,
you got a proper copy of the DJI Battery Killer Tool including the MacOS Version?
Best
Kai

flf3, 2023-12-21:
Not sure.   Do not have the Mac version.  Thanks.

Fil L. Fina III, PE
518-281-8888

Vankay, LLC
Coyote Aviation, LLC
Saratoga Seaplanes, Inc
Stowefall, LLC

Euro Tool Works, LLC
www.uniorusa.com<http://www.uniorusa.com>

QINGHLS, 2025-01-01:
Some batches of the CP2112 have issues, and you may need to connect the battery board to the power input near the CP2112 instead of the GND of the IO port.

rodcanales-hub, 2026-08-15:
Donde lo descargo