o-gs/dji-firmware-tools #372: error 1112

iarkkk9, 2023-12-05:
with this tool I can solve the DJI 1112 activation problem. this problem is linked to the serial number of the drone which is stolen for DJI even if it isn't it doesn't let me activate it. with this tool be able to do it, if so how?
Thanks in advance for your answers

rsrmdp, 2023-12-05:
https://github.com/o-gs/dji-firmware-tools/blob/master/README.md

Wawendgr8, 2024-05-15:
does this work?

andrei8181, 2024-05-18:
Hello everyone! If you have a drone that's locked to another DJI account and you're unable to fly it, we have a solution. We can assist you remotely with activating and unbinding your DJI drone from the previous account. Here is a link where you can learn more about the unlocking process and get our help. telegram @andrewpetcu
https://youtu.be/m1pxbYMCr9w?si=tdQIFnATlIIV7QHX

napHuLLIa, 2024-06-05:
Can you help with activated dji action 4?? I cannot do it, error activated