o-gs/dji-firmware-tools #375: DJI Mini 4 Pro gimbal calibration

estebangmi, 2023-12-15:
I replaced the cable and gimbal on my mini 4 pro after a crash, now I no longer receive " gimbal stuck" messages, but the camera is slightly tilted and facing away. I tried to turn the motor one position but it still looks to the side. I thought about using the comm_og_service_tool.py tool but did not find my model in the list of available ones. Perhaps there is some way to do the correct calibration?

alives2000, 2023-12-15:
I can calibrate the gimbal on this model. PM me in WhatsApp https://wa.me/+6281284309143 or telegram https://t.me/djianderson

flamewow, 2025-02-21:
> I replaced the cable and gimbal on my mini 4 pro after a crash, now I no longer receive " gimbal stuck" messages, but the camera is slightly tilted and facing away. I tried to turn the motor one [...]

Did you figure it out?