o-gs/dji-firmware-tools #381: Getting Error Logs from Mavic Air 2

comsma, 2024-01-19:
I have a Mavic Air 2 with forward vision issues. I recently replaced the gimbal because it was involved in a crash.

I am trying to see if i can pull logs from the drone to figure out if it is an issue with the gimbal not being paired or if the forward vision sensors are damaged.

I have tried using wire shark and the comm_dissector tools to analyze packets from the usb  connection to the drone but I am starting to suspect that this is not the way to listen to the serial interface of the drone.

Would i need to wire a serial connection to the drones main control board to listen to these logs or is there a way to use comm_serial talk to get the statuses of these modules in the drone.

mefistotelis, 2024-01-19:
I did used the dissectors with packets wrapped into USB, as well as WiFi.

For whether they will be helpful with MA2 - I don't know.

comsma, 2024-01-19:
how did you make the usb connection to the drone? Did you connect to the drones on board usb port or did you connect a uart adapter to the drones control board?

mefistotelis, 2024-01-19:
I did used UART connections a lot in old drones, which had more of separate modules.
Now there are less programmable chips inside, so not sure if any such connections remain.

For USB - I used the external one, but in anything above Mavic 1 this requires sending a specific packet within like 5 seconds of power on. Then the USB gets disabled.

The packet can be sent by the original DJI software, the "Assistant" - if it exists for your drone.
Can't remember details; I planned to add that to the service tool, but I don't think I did that finally.

alives2000, 2024-01-22:
You went the wrong way, you can just calibrate the vision sensors in the dji assistant, if one of the sensors is faulty, you will not see the image from this sensor

comsma, 2024-01-22:
> You went the wrong way, you can just calibrate the vision sensors in the dji assistant, if one of the sensors is faulty, you will not see the image from this sensor

I can't get past the initializing screen for the calibration. Would this only become stuck due to the vision sensors or could it throw issues when the gimbal has errors?

alives2000, 2024-01-22:
I think you need to use a different computer, this is not related to problems with the drone

comsma, 2024-01-22:
@alives2000 I've tried this on both windows 11 and osx 14. It is not an issue with the computer. DJI is making it difficult to identify the issue as I get a generic forward vision system error unless that is referring. Does the forward vision system also require the gimbal camera to be functioning or is it referring only to the vision system?