o-gs/dji-firmware-tools #383: trying to clear error 10016 on Air2AS

Judslat, 2024-02-13:
Might be barking up the wrong tree with this but ive been told that running the below command  from the comm_og_service_tool.py tool may clear the error
So I issue the command python comm_og_service_tool.py --port com3 -vv wm232 CameraCalib EncryptCheck 
then run 
python comm_og_service_tool.py --port com3 -vv wm160 CameraCalib EncryptPair 

and i get  Error: Calibration for selected platform is not supported. 

Ive tried changing the WM number to different quads with the same error each time, the Drone is on a FW with ARB5.

Any ideas or is there another fix for error 10016_

rsrmdp, 2024-02-13:
10016, it is an error in the camera sensor. no gimbal calibration

mefistotelis, 2024-02-13:
> then run
> python comm_og_service_tool.py --port com3 -vv wm160 CameraCalib EncryptPair
> and i get Error: Calibration for selected platform is not supported.

Lucky you. Trying to pair encryption post-production would only update keys in chips which were not set to write-protect after the original pairing in factory. The other chips would keep the previous keys, making complete encryption desync.

Judslat, 2024-02-14:
> 10016, it is an error in the camera sensor. no gimbal calibration

Strange how it happened after clearing a 40021 error though.

Oh well looks like no one has an answer to the error

alives2000, 2024-02-15:
To remove error 10016, the first thing - you need to put the drone into service mode (OTA), the second thing - to have a camera calibration file, it's different for each camera module. And several DUML packages to activate the calibration file. The problem is that all these things can only be done with access to the dji vpn account. There is another way - an exploit to get adb root access and manually replace some files

KTM1551, 2024-05-19:
> To remove error 10016, the first thing - you need to put the drone into service mode (OTA), the second thing - to have a camera calibration file, it's different for each camera module. And several [...]

What is the exploit method used to get ADB Root Access to the Air 2s? I have one Air2s on the old FW that is what you downgrade to for the D-H stuff, and i have another Air2s that i was updating until i kind of stopped using it a lot about 1.5-2yrs ago. I got the FPV learned Acro and now i have mostly "diy" CF FPV Drones with Air Unit and Analog..

But ive always wanted to do that, i know its possible, i just havent really looked into it much, ive been studying linux and Android coding for about a year now and would love to just take a look really and see how it works.. 

I have my other Air2s if anything goes wrong, plus my Mini 3 Pro, which unfortunately for videos and photos straight from the Drone, i think the M3P has the advantage as far as that goes. Plus the low light filming is significantly better it seems...

Thanks!

robxs1, 2024-09-10:
@Judslat did you ever get this error resolved ?10016

Judslat, 2024-09-10:
No sadly, it was more complicated than I thought Sent from Android Phone

QINGHLS, 2025-01-01:
> > To remove error 10016, the first thing - you need to put the drone into service mode (OTA), the second thing - to have a camera calibration file, it's different for each camera module. And several DUML packages to activate the calibration file. The problem is that all these things can only be done with access to the dji vpn account. There is another way - an exploit to get adb root access and manually replace some files
> 
> What exploit method is used to get ADB Root Access to the Air 2s? I have one Air2s on the older FW with a low ARB#, and i also have another Air2s that i was periodically updating until i kind of stopped using as much about 2yrs ago now. I got the DJI FPV for a great deal, and I learned Acro on that after a long time of not wanting to try M mode lol..
> 
> Now my newer drones i have are mostly the CF 3" & 5" FPV Drones with DJI Air Unit, O3, and Analog..
> 
> But ive always wanted to get the Adb root access working, i know its possible, i just havent really looked into it much, and when i did i couldn't find too much information about it.
> 
> Ive been studying Linux and Android coding for about a year and a half now and i would love to just take a look at it really and see how it works.. Idk if i have anything to change now, but i do need to replace the front vision sensors on one of them.
> 
> I have my other Air2s if anything goes wrong, plus my Mini 3 Pro, which unfortunately for videos and photos straight from the Drone, i think the M3P has the advantage as far as that goes. Plus the low light filming is significantly better it seems...
> 
> Is it possible to order oem new DJI parts? I bought new Gimbal Mount Rubber Dampers, but after i installed them they were basically as soft as the used rubbers i replaced and one was even ripped, but it seemed better before. It just moves too much..
> 
> Thanks!

Have you implemented adb root? If so, could you tell me how to do it? I can't find relevant information online. 
Thank you very much！

QINGHLS, 2025-01-01:
> To remove error 10016, the first thing - you need to put the drone into service mode (OTA), the second thing - to have a camera calibration file, it's different for each camera module. And several [...]

Have you implemented adb root? If so, could you tell me how to do it? I can't find relevant information online. 
Thank you very much！

alives2000, 2025-11-22:
> Can anyone help me with fixing error 10016 in my DJi Air 2s Only that one error fucked up my mind

Contact me via WhatsApp https://wa.me/+46720169505 or Telegram https://t.me/djianderson

KTM1551, 2026-09-04:
> > Can anyone help me with fixing error 10016 in my DJi Air 2s Only that one error fucked up my mind
> 
> Contact me via WhatsApp https://wa.me/+46720169505 or Telegram https://t.me/djianderson

How involved is it to get a set of used Air 2s front vision sensors and replace that piece, to program it to work properly after? if its involved then i wont bother, but somehow one of the obstable avoidance cameras has a tiny crack inside the Lens and it makes the Drone think theres something on that top left corner where the top left camera is just barely noticeablely damaged. 
I havent bought the part yet as i tend to just disbale to avoidance sensors and got used to that, but it would be great to have for the "Auto Modes" for the Videos & Pictures. That way it doesnt hit something doing a Quickshot or mastershot i forget the naming off the top of my head..

Thanks!

davides15, 2026-09-06:
> How involved is it to get a set of used Air 2s front vision sensors and replace that piece, to program it to work properly after? if its involved then i wont bother, but somehow one of the obstable [...]

Nothing complicated, a set of front sensors costs about $20, after replacement, calibration is performed in the dji assistant program (consumer series)