o-gs/dji-firmware-tools #384: dji mini 4 pro decode with dji_imah_fwsig

HackTeacher, 2024-03-09:
I downloaded the latest firmware of dji mini 4 pro from DDD, and used binwalk to export some .sig files, and then used dji_imah_fwsig.py to decrypt, but it reported an error and analyzed the code and found that the key was wrong, so I want to know how to do it Looking for key? Analyze dji_upgrade and extract it from it?

hackycrash, 2024-09-18:
Decryption keys in `dji_imah_fwsig.py` seems deprecated.

Is there a generic method to get decryption key? @mefistotelis

I am trying to find a way to get decryption key for osmo action 4 firmware. 

If you let me know, i will do it and contribute to this project.

mefistotelis, 2024-09-18:
> Is there a generic method to get decryption key?

No. The methods changed between devices, as DJI tried to patch all the holes in their implementation.

The remaining methods are not public, and can be published only by the people who extracted the keys.

Breaking into the ARM TrustZone does require studying the subject in detail.

When we were starting, it was enough to create a `.tar` file which stores file name like ` | cat cat /proc/meminfo` and using it as FW update file. The DJI update procedure would extract the TAR, and execute a command on each file inside - executing the command provided. Then we used race condition while starting two FW updates at the same time, and start installation of 2nd FW after passing verification of only the 1st one. In the meantime we also used storing symbolic links is TAR files, and few other techniques. When TrustZone was introduced, we used some holes DJI made to workaround their lack of understanding of its rules - some special commands. On other platforms we used various techniques to capture encrypted binaries, starting at bootloader, then analyzed them to find a hole in next layer. Sometimes DJI was also publishing stuff they did not intended to publish, like giving drones with early pre-release FW to influencers, or just publishing FW or other files with some extra presents inside.

KTM1551, 2024-09-28:
Yea i tried recently to do this same thing and got as far as the.sif files.

I was working with an Avata FW file. 

The tool to Extract the files from the .sig files doesn't seem to work on these newer drones like stated above.

I would love to try something easier first, and then try the DJI FW again lol..

It's definitely not something you can just research on your own and learn ive found out pretty quickly. I dont have any training in coding like this..

Synmac05, 2025-07-24:
I've encountered the same problem too. Any new progress now?