o-gs/dji-firmware-tools #388: DJI Air 3

Ric-1000, 2024-04-02:
DJI Air 3 is obviously not supported yet. When will it be supported, or can I use a different product code for gimbal calibration?

Michael515613, 2024-04-02:
> DJI Air 3 显然还不支持。何时支持它，或者我可以使用不同的产品代码进行云台校准？

I can do it.

Michael515613, 2024-04-02:
t.me/Michael_Yb

kramer123, 2024-04-02:
Do what,  I don't remember what I posted. 

 

 
 

Sent: Tuesday, April 02, 2024 at 8:25 AM

alives2000, 2024-04-03:
> DJI Air 3 is obviously not supported yet. When will it be supported, or can I use a different product code for gimbal calibration?

I can calibrate the gimbal on this model remotely. Write to me in telegram https://t.me/djianderson

israeltivon, 2025-10-20:
> > ברור ש-DJI Air 3 עדיין לא נתמך. מתי הוא יתמוך, או שאוכל להשתמש בקוד מוצר אחר לכיול גימבל?
> 
> אני יכול לכייל את הגימבל בדגם הזה מרחוק. כתבו לי בטלגרם https://t.me/djianderson

What is your email or WhatsApp? I need help calibrating the AIR3 gimbal.