o-gs/dji-firmware-tools #392: Dji mini 4 pro 1D020001 error

davides15, 2024-04-30:
[image]

Does anyone know how to get rid of the gimbal calibration error on the mini 4 pro? The gimbal works perfectly but the error remains. I tried different paid firmwares from DDD app, it seems that they all don't work

alives2000, 2024-04-30:
I know how to do it for this model and for air 3 too. Text me in WhatsApp https://wa.me/+6282172769186 or in telegram https://t.me/djianderson

davides15, 2024-04-30:
Why do I have this error if the gimbal is working good?

alives2000, 2024-04-30:
Gimbal calibration error 1D020001 or 40011 always appears after the factory gimbal calibration

RICKYMCHELSEA, 2025-05-15:
How can I calibrate my mini 3 pro gimbal after calibration