o-gs/dji-firmware-tools #396: Mavic Air 2 with errors 40021 and 10017 after camera replacement

SkyMaster1, 2024-05-25:
Hello
The camera in my Mavic Air 2 burned out and the gimbal also didn't work. After I installed a new camera, almost everything works again, except the zoom doesn't work and I get the two errors 10017 and 40021.
is there a way to delete these errors?
I still have the original camera in case the serial number is needed.
Thanks
Best SkyM

kramer123, 2024-05-25:
Yes,  first put the original camera in there, that will eliminate one error. 

if you ran the python calibration ,  you can run the service firmware from dank download to fix it. 

i think the whole process is on youtube

 
 

Sent: Saturday, May 25, 2024 at 10:57 AM

SkyMaster1, 2024-05-25:
As written, the original camera no longer responds. 
Whether I install the defective original camera or no camera at all, I don't notice any difference.
I haven't found anything yet on, how I can replace the defective camera, without the two error codes appearing.

> Ja, zuerst die Originalkamera hineingesetzt, die einen Fehler eliminiert. Wenn Sie die Python-Kalibrierung ausführen, können Sie die Service-Firmware von einem feuchten Download ausführen, um sie zu beheben. Ich denke, der ganze Prozess ist auf Youtube Gesendet: Samstag, 25. Mai 2024 um 10:57 Uhr Aus: "SkyMaster1" ******** Zu: "o-gs/dji-firmware-tools" ******.*** Cc: "Abonniert" ******** Betreff: [o-gs/dji-firmware-tools] Mavic Air 2 mit Fehlern 40021 und 10017 nach Kamerawechseln (Ausgabe [Nr.](https://github.com/o-gs/dji-firmware-tools/issues/396) 396) Hello Die Kamera in meinem Mavic Air 2 brannte aus und der Gimbal funktionierte auch nicht. Nachdem ich eine neue Kamera installiert habe, funktioniert fast alles wieder, außer dass der Zoom nicht funktioniert und ich die beiden Fehler 10017 und 40021 bekomme. Gibt es eine Möglichkeit, diese Fehler zu löschen? Ich habe immer noch die Originalkamera, falls die Seriennummer benötigt wird. Dank Beste SkyM - Klicken Sie auf diese E-Mail direkt, sehen Sie sie auf GitHub ein oder melden Sie sich ab. Sie erhalten dies, weil Sie diesen Thread abonniert haben. Meldungs-ID: ******.**

SkyMaster1, 2024-05-28:
Is there a way to pair the new camera with the drone? Maybe that already solves the problem.

sonyox, 2024-08-16:
[image]

kramer123, 2024-08-16:
You are lucky the drone is asking to be activated and failing, if it works be happy.   there probably is a way to change the serial # and get rid of 10017 error,  the 40021 error you'll need to flash it with dank downloaded service FW,  cost about $2.  I can do it remotely orlando drone repair com