o-gs/dji-firmware-tools #400: Chip detection failded: error when conecting a laptopbatery toshiba model l800

skysurf, 2024-07-11:
./comm_sbs_bqctrl.py -vvv --dev_address 0x0b read BatteryStatus
Opening i2c:1
Raw write: DATA=22 3e
Retrying due to error: ConnectionError: FW version acquire tripped as it expected NACK on a command
Raw write: DATA=22 3e
Retrying due to error: ConnectionError: FW version acquire tripped as it expected NACK on a command
Raw write: DATA=22 3e
Chip detection failded: ConnectionError: FW version acquire tripped as it expected NACK on a command
Auto-selected chip: SBS, Generic chip with SBS support
Warning: Could not open chip definition file 'comm_sbs_chips/SBS.py'
Reading simple command at addr=0xb, cmd=0x16, type=uint16, opts={'subcmd': None}
Raw BatteryStatus response: ff ff ff
Retrying due to error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum
Raw BatteryStatus response: ff ff ff
Retrying due to error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum
Raw BatteryStatus response: ff ff ff
Error: ValueError: Received uint16 from command BatteryStatus with wrong PEC checksum

i have this error when i try to read from a laptop batery toshiba model l800

mefistotelis, 2024-07-11:
Auto-detect errors: The detection feature was implemented for TI BQ-series chips. If you have a different chip, you can tell the tool to use generic chip - there's cmdline option for that.

For `Raw BatteryStatus response: ff ff ff` - no idea. I guess you'd have to check documentation of your chip. Maybe it won't talk until unsealed, maybe that means a specific thing. The response should be 3 bytes long, I think - 16-bit value and 8-bit checksum.

skysurf, 2024-07-13:
hi 
this battery works with bq9000 TI chip, this is the chip : https://i.imgur.com/W4poRJV.jpeg