o-gs/dji-firmware-tools #403: DJI Phantom 4 Advanced no reply

KenwoodFox, 2024-08-23:
Im trying to recalibrate my hall sensors after a motor replacement but, im not getting a reply back from the drone, my model is `WM332A`

its likely not a bug with the software but any help would be appreciated! Thanks! Heres what i see on my end

```
❯ python comm_og_service_tool.py --port /dev/ttyACM0 -vv wm332 GimbalCalib LinearHall
Opened /dev/ttyACM0 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 ce c6 20 04 08 02 ff 57
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 54 packets (4339b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 ce c6 20 04 08 02 ff 57
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 61 packets (4739b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 ce c6 20 04 08 02 ff 57
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 54 packets (4350b), dropped 0 fragments (0b)
No response received.
Error: No response on calibration command LinearHall request.
```

[image]

mefistotelis, 2024-08-23:
The auto-calibration function was introduced in Mavic.

Ph4 had half-manual calibration, and we never developed a tool to perform it.

Can't remember but there are probably some threads about how repair workshops deal with that, try searching.

KenwoodFox, 2024-08-23:
Oh! Yeah, i found a few threads about replacing all the magnets/swapping the motor housings. If you can buy motor housings online, surely you can calibrate them right?

Looks like its off-topic for this repo though :c sorry, thanks for the advice. I think im just going to have to live with it being cockeyed unless i can adjust all the magnets

closed 2024-08-23 by KenwoodFox

KenwoodFox, 2024-08-23:
Ah, looks like this was already brought up in #99 

Man, this kinda sucks.. Even if i swap the magnets around I'll never be able to center the camera exactly, what a terrible solution