o-gs/dji-firmware-tools #407: Mini 3 batteries won't unlock.

and-sh, 2024-09-30:
Hi gangsters
Can not unlock mini3 batteries. Two pieces. One original, one not original. Was something changed recently?
[image]
[image]

and-sh, 2024-09-30:
Besides
Is there some logger and decoder for dji messages?
I have all wires accessibly in my setup, so I can peek a little.
[image]

KTM1551, 2024-10-01:
> Besides Is there some logger and decoder for dji messages? I have all wires accessibly in my setup, so I can peek a little. ![m2p](https://private-user-images.githubusercontent.com/43984737/372116284-44fc3430-889f-4db8-84ca-e6d698cb2a1d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc3NzUwNTQsIm5iZiI6MTcyNzc3NDc1NCwicGF0aCI6Ii80Mzk4NDczNy8zNzIxMTYyODQtNDRmYzM0MzAtODg5Zi00ZGI4LTg0Y2EtZTZkNjk4Y2IyYTFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAxVDA5MjU1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmMjEzZWZiNTA2ZGY0N2RmMGM5ZjBkNDA1MTBmZWUyYzU5OGQzNzJmYzMxZWM5N2EwYWVjMzlmYmVmZDBjOTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.5nXqznyFSyrgTvjFLFcOa6i9-ZyMNsOjwL1jpxzz3zI)

Thats awesome! So youre running 2x18650/21700 Liion cells with the stock BMS or did you find one to use in its place?
Does it still behave like a stock battery in the App and on the Drone for RTH and all that? Cool work! I love my Mini 3 Pro!!

How is the run time with this setup vs the stock battery?

and-sh, 2024-10-01:
Thanks
2x 21700 6000mAh vapcell, total weight about 150g
BMS from not original battery plus. It behave like stock battery.
Drone weight was slightly reduced. With these batteries it is roughly the same as a standard drone with a plus battery.
Flight time is not satisfactory, it is about 40 minutes to 3V per cell. Due to large sag I can use only about 4.6Ah from battery.  And power capacity (Wh) is lower than with LiPo battery.
Typical discharge curves [here](https://budgetlightforum.com/t/bench-test-results-vapcell-f60-12-5a-6000mah-21700/222498

Looking for alternative batteries.

[image]
)

mililanigal, 2024-11-09:
Did you recently run a firmware update?  I did, and the drone now detects the batteries as non DJI batteries.  Also, I was planning on doing a custom battery like you with high discharge (25A) samsung cells -- around same capacity.  But, I'm really surprised your flight time is still about the same as the flight plus battery, with almost half the battery capacity and even more weight.  It makes me wonder if the advertised battery capacities are not accurate.

and-sh, 2024-11-11:
Yep, dji began to actively fight with clients.
Recently I downgrade firmware to .600. It seemed to me that video transmission works better on it.
Battery plus seems to be very well balanced, you can win only if you use very good batteries.
Molicel p50b is in my opinion, more promising.
To resolder the BMS I drove it into Shutdown mode with the help of dji battery killer. Disolder plus terminal first, solder back last. To start bms short positive terminal with output.
But I do nor recommend to modify drone
if you've played with it enough, it's easier to sell it and get an prebuild drone with inav. 
If you want hi quality video use gopro and gimbal. 
HD fpv - Caddx has some good solutions. They have no problems switching to FCC.
With avatar hd vrx you can use any monitor or hdmi headset.
For Long range - use 1.3Ghz analog VTX and 433 tx like myflydream. Yoy can go up to 50+km this way.
Very promissing project is openhdfpv. If they get popular, China will make inexpensive kits for them like FC for Betaflight.
Progress does not stand still, with such an attitude towards clients, they will forget about DJI.

mililanigal, 2024-11-11:
Much thanks for the response.  I'm guessing you're on a Mini 3 Pro?  I'm on the Mini 4 Pro, and I don't think anyone has a way to crack it.  I unfortunately can't downgrade the firmware.  So, you're right, it would be better off to sell it and build my own.  I'm still a big fan of DJI's air system, and their microOLED goggles.  They are release the standalone O4 air unit, so I might just get that and build my own.  

One final thing, have you thought about running another dji battery in parallel with the internal battery?  That is, solder another DJI battery to some connectors and connect them in parallel.  That should work with identical batteries, and effectively double the capacity.  You will have the second battery extend underneath the drone, but for you, since you cut out the bottom already, should be fine.  You can just velcro it to the undercarriage.  I'm thinking of doing something like that.

christhomas, 2024-12-13:
I also have the mini 4 and a "non dji battery" like you @mililanigal and what I read is that apparently dji changed the key used to unlock the batteries, so apparently the methods people use on the dji mini 3 aren't working yet. So I was waiting for a while to see whether somebody had found a way to find what this key is and then I have a CP2112 breakout board and try to wire it up. 

Also I build FPV drones too, the problem with them is that they never seem to be even close to the reliability of DJI drones. I wish it was true to build an FPV drone that can rival DJI. But the software is just years behind how advanced DJI software is. Return to home functionality does exist in betaflight, but it's unreliable. Sometimes it works, sometimes it doesn't, sometimes it limps home and it's like the drone doesn't know how to fly itself straight, othertimes people have great experiences. But this kind of behaviour makes me really nervous about using it cause you never know what you're gonna get.

These experiences might be because the hardware is just low quality components. In FPV we buy flight controllers and ESC boards and we aren't really sure whether we're getting top quality components or garbage. So when the software runes perhaps it outputs rubbish data. Or perhaps the way you wired the components creates electrical noise so the data is being corrupted. Perhaps you bridged some wires at the soldering pad and it touches, but perhaps the wire vibrates and it touches sometimes. 

Basically, stuff happens when you build things yourself that you can't match DJI build quality with their automated machinery. Your quality probably is lower than their professional factory assemblies. So the FPV result is always a kind of "probably it's a good job if you're skilled"... but how many of us are really amazing at this? Some people are. I think most people aren't. I've even seen joshua bardwell soldering and it's not great to be honest. I've done better and he's one of the best known FPV pilots in the hobby. 

What I'm hoping is that we can learn how to get around the restrictions DJI put on the batteries, so I can hook up my battery and tinker around with it and unlock it, unbrick it, maybe make it so DJI can't detect it's not a third party battery anymore. Maybe they are able to detect what chips are used in the board and when it doesn't match the original signature thats how it knows. Perhaps the IC has the wrong part number, so thats how. Perhaps we can "lie" to dji about the part numbers, or serial number or something in order to get the battery back to working order again.

and-sh, 2024-12-13:
You may want to use Pixhawk it has double redundancy.

mililanigal, 2024-12-13:
> I also have the mini 4 and a "non dji battery" like you @mililanigal and what I read is that apparently dji changed the key used to unlock the batteries, so apparently the methods people use on the [...]

Have you thought about cannibalizing the BMS from an old oem DJI battery on the non - oem?  That's what I plan on doing.  You can find some pretty cheap OEM batteries that were used with under 100 cycles.  I bought one off of Ebay from gotdrones for $20.  I plan on "upgrading" that battery with the cells from the non-OEM battery.  It will get a bigger capacity.  I just need to program it for the correct capacity.  

As for the keys to unlock the battery, maybe they changed the key for the newer batteries, but the old ones seem to unseal fine.  The thing you can't seem to do is change the battery info on the BMS, and that's what I'm thinking DJI has done to detect whether the battery is OEM or not.  Maybe via the serial numbers, I'm not really sure, but I cannot modify the battery info parameters via the software, even on the non-OEM batteries.  I believe these are permanently coded into the firmware.  Unless someone is able to correct me.

and-sh, 2024-12-14:
It is a good idea
I can guess that new firmware change key at firt connect. If you have old battery you can check it. 
Dont forget to put bms in shutdown mode before resoldering. In shurdown mode bms does not respond to battery killer.

mililanigal, 2024-12-15:
> It is a good idea I can guess that new firmware change key at firt connect. If you have old battery you can check it. Dont forget to put bms in shutdown mode before resoldering. In shurdown mode bms does not respond to battery killer.

Hmmm, that's interesting.  I've never heard of that before.  What is the reason for shutting down the BMS before de-soldering?  I haven't seen or heard of anyone doing this before.

and-sh, 2024-12-15:
It will go to PF state with good probability. If you know code you can clear PF after soldering. If no, you can use this bug.