o-gs/dji-firmware-tools #408: amba_fwpack fails with Error: Detection finished with unusual entry sizes, verify files

mprevot, 2024-10-04:
Trying to unpack firmware from gopro session 5 / 5 black (publicly available) with `python3 amba_fwpak.py -m camera_loaders.bin -x`, but it fails with 

>  Error: Detection finished with unusual entry sizes, verify files

This an ambarella A9SE7 (https://github.com/hypoxic/GoPro-Research), so I thought this should work.

If not, how to adapt or correct amba_fwpack.py ? Any hints ?

mefistotelis, 2024-10-04:
Ambarella provides SDK for device makers, which they modify to create SW for their products.

If the tool doesn't recognize the specific container, use search option to just recognize partitions - the partition markers never change.

mprevot, 2024-10-04:
Unfortunately, amba_fwpak.py fails to find the pattern `epos = fwmdlmm.find(b'\x90\xEB\x24\xA3'` and exits silently.

https://github.com/o-gs/dji-firmware-tools/blob/156a1f95b3e254cdf8fd577d31279017c300b8c6/amba_fwpak.py#L547

fw for repro:

https://gopro.com/en/us/support/hero5-session-product-update/windows
or
https://1drv.ms/u/c/d9842e22f46d2655/Ea7woPSqngpHj7s4uRQN-FsBXzNBmrs-jCT2I8wNSDF1dg?e=Xibwvg

mefistotelis, 2024-10-05:
> Unfortunately, amba_fwpak.py fails to find the pattern

Not sure what to tell you..:

```
# ./amba_fwpak.py --fwmdlfile camera_firmware.bin -s
camera_firmware.bin: Extracting entry  0, pos      236, len   234222 bytes
camera_firmware.bin: Extracting entry  1, pos   234714, len    11176 bytes
camera_firmware.bin: Extracting entry  2, pos   246146, len  7999488 bytes
camera_firmware.bin: Extracting entry  3, pos  8245890, len 45473248 bytes
camera_firmware.bin: Extracting entry  4, pos 53719394, len 11026432 bytes
camera_firmware.bin: Extracting entry  5, pos 64746082, len  5669000 bytes
camera_firmware.bin: Extracting entry  6, pos 70415338, len 35913728 bytes
```

That's on file from the link [hero5-session-product-update/windows](https://gopro.com/en/us/support/hero5-session-product-update/windows).

Some header fields are clearly incorrect - header format is different. But the partitions are extracted.