o-gs/dji-firmware-tools #416: Any experience with Mini 3/4 Pro aftermarket batteries based on SH366000 chip?

pawelsky, 2025-01-15:
Hi does anyone (@mefistotelis? @mixeysan?) have any experience with Mini 3/4 Pro aftermarket batteries based on SH366000 chip that imitates bq40z307?

Couple of months ago a firmware update made these batteries 'unflyable' which is a pity. I've done some experiments and managed to figure out that (as it happened in the past) the aftermarket battery serial numbers are blacklisted, and when the serial number (the one that can be read with D8 command) is replaced with a known good one during the I2C communication, then the battery is considered good again by the firmware.

The problem is that I'm unable to replace to permanently replace the number in the battery itself as it seems to be sealed. Tried some known unseal keys but with no luck. The battery firmware seems to imitate the bq40z307 firmware well enough, but no luck with unsealing (tried using both the standard bq40z307 unseal commands as well as SH366000 specific ones (using [this tool](https://github.com/lmdpua/CP2112_SH366000_Flasher)))

Did anybody have any experience with the SH366000 chip before?

Here is the the battery itself

[image]

And here are the internals (not the exact same battery, image borrowed from someone else's teardown).

[image]

Here is the basic datasheet for SH366000 (in Chinese but can be easily Google-translated)

[SH366000 User Guide CV0.6.pdf](https://github.com/user-attachments/files/18430636/SH366000.User.Guide.CV0.6.pdf)

mefistotelis, 2025-01-15:
Unsealing method is clearly different - 2x 16-bit number is the key/password in original, in the clone you have 4x 16-bit number. You also initiate the unseal differently - in original you just send the password (which means the password cannot be the same as existing commands) and it works. Here, you send 0x08FF and then the password (in following 8 bytes).

The implementation in the unofficial tool is here:
https://github.com/lmdpua/CP2112_SH366000_Flasher/blob/master/mainwindow.cpp#L1088

Anyway, I don't have any experience with this chip.

Have you tried "typical value" from the spec? "0xaa 0xbb 0xcc 0xdd"? Also worth checking simple values following the same scheme as default key, like "12 34 56 78".

pawelsky, 2025-01-15:
> Unsealing method is clearly different - 2x 16-bit number is the key/password in original, in the clone you have 4x 16-bit number.

True, but it also imitates a lot of functionality and messages of the bq40z307 so as mentioned I've tired it both the bq40z307 unseal as well as the SH366000 one

> Have you tried "typical value" from the spec? "0xaa 0xbb 0xcc 0xdd"? Also worth checking simple values following the same scheme as default key, like "12 34 56 78".

Yes, I've tried a couple of 'possible' passwords, but none of them worked.

BTW The SH366000 unseal method requires words not bytes, but I've tried "0xaaaa 0xbbbb 0xcccc 0xdddd", "0x6161 0x6262 0x6363 0x6464" (ASCII for aa bb cc dd from the PDF), "0x4141 0x4242 0x4343 0x4444" (ASCII of the uppercase variant AA BB CC DD), "0x0123 0x4567 0x89AB 0xCDEF" (because why not), various combinations of the known bq40z307 passwords (unseal and full access ones combined have the length required) and some other random ones that came to my mind.

pawelsky, 2025-04-24:
Hi @0r10nV The firmware does support some limited set of bq40z307 opcodes but not many. As mentioned I've tried both bq400307 and sh366000 native unseal methods but no luck. I could not find a working unseal keys but if you know any that could be tested I'm all ears :)

0r10nV, 2025-04-25:
me dont have unseal key for this battery. 
there is also probability that unsealing this chip will not grant access to DJI_BARCODE parameter location, because it is handled by non-standard register 0xD8 which is outside of SBS spec.
 As you could know SH36 is SBS-compatible chip so it handles only the registers set from 0x00 to 0x23 described in standard. Additionally it could handle CellsVolts registers 0x3F...0x3C outside the standard range and some security stuff. 
 But DJI uses a lot of their proprietary stuff in bq40z307 like ECC, MAC, DRM, BFU and so on. It could not be handled by native SH36 firmware so manufacturer installed extra FPGA assembly on main board. It is located in the right side of the PCB picture you provided. This is where all this Dji extra stuff could be located.

pawelsky, 2025-04-25:
> there is also probability that unsealing this chip will not grant access to DJI_BARCODE parameter location, because it is handled by non-standard register 0xD8 which is outside of SBS spec.

0xD8 is supported, but not writeable :(

> It could not be handled by native SH36 firmware so manufacturer installed extra FPGA assembly soldered on main board. It is placed in the right side of the PCB picture you provided.

Interesting, a path worth exploring (that unfortunately means I'll have to cut open the battery case - something I've really wanted to avoid).

Do you happen to know what FPGA chip could that be (you don't see many in such a small package)?

pawelsky, 2025-04-25:
Got access to the chip, unfortunately cannot see any markings through the epoxy, only something resembling H6 or 9H on the (I guess) LDO chip in the HSNT-4 package on the left.

[image]

0r10nV, 2025-04-25:
> 0xD8 is supported,

This register along with others like 0x66, 0xC0, 0xD9 0xDA used by Dji could be handled by another chip not by sh36 directly. For example 2 smbus slaves sitting on same bus. SH36 handles SBS range and other chip responds to outspec registers. SinoWealth produces SH79 chip also for developers which is blank version of SH36. This is where designers could implement any sort of proprietary stuff by developing firmware from the scratch. SH79 is very popular in laptop copy batteries. But for some reasons preprogrammed version was instead which makes me think Dji stuff handled separately outside.

0r10nV, 2025-04-25:
> Do you happen to know what FPGA chip could that be

No. But several years ago it was teardown report of Dji Mavic Pro copy battery made by PowerPlant. It also used similar submodule soldered on main pcb, but chip marking text was well readable, and someone googled it and found it was FPGA ic.

pawelsky, 2025-04-26:
Managed to read the markings on the chip by putting a drop of IPA on top of the resin to smooth out the surface.
It does indeed seem to be an FPGA module (**UL1K36A** - [Lattice iCE40 UltraLite  UL1K](https://www.latticesemi.com/en/Products/FPGAandCPLD/iCE40Ultra))

[image]

Any experience programming these anyone? :)

0r10nV, 2025-04-26:
Chinese guys who replicate tons of laptop batteries have the best programming experience of these ICs although )

pawelsky, 2025-05-10:
Desoldered the chip and dissolved the epoxy to expose the extra pads. WIll try to connect a programmer and see what I can do with it.

[image]

0r10nV, 2025-05-10:
Which programmer do you plan to use with this Lattice Ic ?

pawelsky, 2025-05-10:
Plan is to start with FT232H and take it from there.

pawelsky, 2025-05-10:
Interesting observation - even with the FPGA module removed the SH366000 still returns the DJI specific serial number (the one retrieved with the 0xD8 command)...

0r10nV, 2025-05-10:
So they reflashed it with custom firmware. Have assumed that blank sh79xxx would be more convenient for such tasks.

pawelsky, 2025-05-10:
This begs the question - what is the FPGA module used for?

0r10nV, 2025-05-11:
to compute correct ECC response based on the random challenge received from drone. Otherwise battery considered like not genuine and throws the message "Authentication failure" in the app which does not allow to take off the drone.