o-gs/dji-firmware-tools #417: DJI Air 3 Battery pinout

GekoCH, 2025-01-27:
hy

can someone help me with the DJI Air 3 Battery pinout, since there are now 9 connections and not the 10 I'm used to.
Thx

RudraSama, 2025-03-31:
[image]

GekoCH, 2025-04-01:
perfect thx

closed 2025-04-01 by GekoCH

RudraSama, 2025-04-01:
> perfect thx

But there is no tool for fixing Air 3 batteries (DJI Battery Killer doesn't with A3), then whats the of use pinout?

GekoCH, 2025-04-06:
> > perfect thx
> 
> But there is no tool for fixing Air 3 batteries (DJI Battery Killer doesn't with A3), then whats the of use pinout?

I just use it to read out the battery.
But i still havent found any air 3 battery connector (male) for the battery.
Have you found them?

RudraSama, 2025-04-06:
> > > perfect thx
> > 
> > 
> > But there is no tool for fixing Air 3 batteries (DJI Battery Killer doesn't with A3), then whats the of use pinout?
> 
> I just use it to read out the battery. But i still havent found any air 3 battery connector (male) for the battery. Have you found them?

I use Jumper wires lol