o-gs/dji-firmware-tools #440: HP SB03XL battery with BQ30z55 - Can't unseal - PEC Checksum Errors, but partially works disabling the checks ?

deividAlfa, 2025-07-13:
Hi! I'm trying to unseal this battery.

First I manually charged the cells in series with 12V @ 500mA, they charged slowly and very well balanced, with less than 50mV between cells.

The SBS is dead unless I apply 12V externally.

I used a small arduino board as USB-I2C bridge (Thanks, @Cleric-K !): https://github.com/o-gs/dji-firmware-tools/issues/243#issuecomment-3005323944

But, I only received `wrong PEC checksum`.

I used a logic analyzer to make sure teh data was correct. And indeed it was. For example:

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z55 read ManufacturerAccess.Voltages`

The captured response frame (Including dev. address) was:

    17 0C E4 0E F6 0E 00 00 00 00 4E 1C 8D 31 09

CRC-8-CCITT matches the last crc byte, `09`, but the first CRC seems to be wrong or use a different algo.

I bypassed all `raise ValueError("Received {} from command {} with wrong PEC checksum"` errors, and the output looked quite meaningful:

    Opening i2c:1
    Serial port COM6 opened
    Importing comm_sbs_chips/BQ30z554.py
    Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ30.Voltages: 113>}
    Query ManufacturerAccess.Voltages: 00 WORD=0x71
    Write ManufacturerAccess: CMD=00 WORD=71 00
    Raw ManufacturerAccess.Voltages response: 0c e4 0e f6 0e 00 00 00 00 4e 1c 8d 31 09
    MA.Voltages:    hex:e40ef60e000000004e1c8d31    struct  Outputs voltage data values
     CellVoltage0:  3812    mV      Cell Voltage 0
     CellVoltage1:  3830    mV      Cell Voltage 1
     CellVoltage2:  0       mV      Cell Voltage 2
     CellVoltage3:  0       mV      Cell Voltage 3
       BATVoltage:  7246    mV      BAT Voltage
      PACKVoltage:  1268500 mV      PACK Voltage

Readings on Cell2, Cell3 and PACK looks strange...
This is a 3S battery, I've manually measured around 3.8V on each cell.

If I remove external power, I have few seconds before the SBS turns off, Pack voltage reads normal in this instant:

    Raw ManufacturerAccess.Voltages response: 0c e3 0e f2 0e 00 00 00 00 2a 1c 7b 00 ca
    MA.Voltages:    hex:e30ef20e000000002a1c7b00    struct  Outputs voltage data values
     CellVoltage0:  3811    mV      Cell Voltage 0
     CellVoltage1:  3826    mV      Cell Voltage 1
     CellVoltage2:  0       mV      Cell Voltage 2
     CellVoltage3:  0       mV      Cell Voltage 3
       BATVoltage:  7210    mV      BAT Voltage
      PACKVoltage:  12300   mV      PACK Voltage

___

The battery only has 422 cycles, which is not that much!

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z55 read CycleCount`

    Raw CycleCount response: a6 01 d5
    CycleCount:     422     cycles  Number of cycles the battery has experienced

___

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z55 monitor BQStatusBitsMA`

    Raw ManufacturerAccess.SafetyAlert response: 04 00 00 00 00 ce
    MA.SafetyAlert:         0x00000000      bitfields       Safety Alert bits
          CELL_UNDERVOLTAGE:        0=Inactive      [CUV]   Cell Undervoltage
           CELL_OVERVOLTAGE:        0=Inactive      [COV]   Cell Overvoltage
      OVERCURRENT_CHG_TIER1:        0=Inactive      [OCC1]  Overcurrent in Charge 1st Tier
      OVERCURRENT_CHG_TIER2:        0=Inactive      [OCC2]  Overcurrent in Charge 2nd Tier
[... 185 lines cut ...]
     CAL_ADC_CC_ON_MD:      0=Disabled      [CAL]   CAL ADC or CC output on ManufacturerData()

___

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z55 sealing Unseal`

The output is different everytime, here're few more :

    Raw ManufacturerAccess.UnSealDevice response: 02 96 02 42 42 42 42 42 42 42 42 42 42 42 42 42 42 42 42 42 42 42
    Raw ManufacturerAccess.UnSealDevice response: 02 1d 8d c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7 c7
    Raw ManufacturerAccess.UnSealDevice response: 02 d8 b1 df df df df df df df df df df df df df df df df df df df
    Raw ManufacturerAccess.UnSealDevice response: 02 03 4e 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01
    Raw ManufacturerAccess.UnSealDevice response: 02 97 03 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50
    Raw ManufacturerAccess.UnSealDevice response: 02 02 af bd bd bd bd bd bd bd bd bd bd bd bd bd bd bd bd bd bd bd

___

This makes nothing:

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z55  trigger ManufacturerAccess.PermanentFailDataReset`

    Writing write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, v=b'', opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ30.PermanentFailDataReset: 41>}
    Store ManufacturerAccess.PermanentFailDataReset: 00 WORD=0x29
    Write ManufacturerAccess: CMD=00 WORD=29 00
    MA.PermanentFailDataReset:      trigger SUCCESS Trigger switch write accepted

mefistotelis, 2025-07-13:
Here's BQ30z55 Technical Reference chapter:

[image]

As to why your response does not meet the specification - I suspect you can think of the same possible reasons I can.

deividAlfa, 2025-07-13:
I disabled the checks, here's the output:

`python comm_sbs_bqctrl.py -vvv --dev_address 0xb --chip BQ30z50 sealing Unseal`

    Opening i2c:1
    Serial port COM6 opened
    Importing comm_sbs_chips/BQ30z554.py
    Write ManufacturerAccess: CMD=00 WORD=31 00
    Raw ManufacturerAccess.UnSealDevice response: 04 b5 fe b2 d5 18 18 18 18 18 18 18 18 18 18 18 18 18 18 18 18 18
    Prepared ManufacturerAccess.UnSealDevice B1=0123456789abcdeffedcba9876543210d5b2feb5
    Computed ManufacturerAccess.UnSealDevice HMAC1=809722d33cde3ac9fe4d615bae925b94d8d0c1bb
    Computed ManufacturerAccess.UnSealDevice HMAC2=9a3d12e4c88129e02f7a153a438d739d9e279377
    Write ManufacturerAccess.UnSealDevice: CMD=2f BLOCK=77 93 27 9e 9d 73 8d 43 3a 15 7a 2f e0 29 81 c8 e4 12 3d 9a
    Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ30.OperationStatus: 84>}
    Query ManufacturerAccess.OperationStatus: 00 WORD=0x54
    Write ManufacturerAccess: CMD=00 WORD=54 00
    Raw ManufacturerAccess.OperationStatus response: 04 21 7b 50 00 91
    MA.OperationStatus:     0x00507b21      bitfields       Operational Status bits
          SYS_PRESENT_LOW:  1=Active        [PRES]  System present input state low
           DSG_FET_STATUS:  0=Inactive      [DSG]   DSG FET status
           CHG_FET_STATUS:  0=Inactive      [CHG]   CHG FET Status
          PCHG_FET_STATUS:  0=Inactive      [PCHG]  PCHG FET Status
          GPOD_FET_STATUS:  0=Inactive      [GPOD]  GPOD FET Status
              FUSE_STATUS:  1=Active        [FUSE]  FUSE input status
           CELL_BALANCING:  0=Inactive      [CB]    Cell Balancing
               LED_ENABLE:  0=Inactive      [LED]   LED Enable
            SECURITY_MODE:  3=Sealed        [SEC]   Security Mode
           CAL_RAW_ADC_CC:  0=Inactive      [CAL]   Calibration Raw ADC/CC output active
            SAFETY_STATUS:  1=Active        [SS]    Safety Status
        PERMANENT_FAILURE:  1=Active        [PF]    Permanent Failure
     DISCHARGING_DISABLED:  1=Active        [XDSG]  Discharging Disabled
        CHARGING_DISABLED:  1=Active        [XCHG]  Charging Disabled
               SLEEP_MODE:  0=Inactive      [SLEEP] Sleep mode condition met
           SHUTDOWN_BY_MA:  0=Inactive      [SDM]   Shutdown activated by ManufacturerAccess()
          SHIP_MODE_BY_MA:  0=Inactive      [SHPM]  SHIP mode activated with ManufacturerAccess()
             AUTH_ONGOING:  0=Inactive      [AUTH]  Authentication ongoing
        AFE_WATCHDOG_FAIL:  0=Inactive      [AWD]   AFE Watchdog failure
        FAST_VOLTAGE_SAMP:  1=Active        [FVS]   Fast Voltage Sampling
        RAW_ADC_CC_OUTPUT:  0=Inactive      [CALO]  Raw ADC/CC offset calibration output
      SHUTDOWN_BY_VOLTAGE:  1=Active        [SDV]   SHUTDOWN activated by voltage
              SLEEP_BY_MA:  0=Inactive      [SLEPM] SLEEP mode activated by ManufacturerAccess()
         INIT_AFTER_RESET:  0=Inactive      [INIT]  Initialization after full reset
           SMB_CAL_ON_LOW:  0=Cal starts    [SLCAL] Auto CC offset calibration on low
     QMAX_UPDATE_IN_SLEEP:  0=Inactive      [SLEPQM]        QMax update in SLEEP mode
     CURRENT_CHK_IN_SLEEP:  0=Inactive      [SLEPC] Checking current in SLEEP mode
         XLOW_SPEED_STATE:  0=Inactive      [XLSBS] Fast Mode

Cleric-K, 2025-07-14:
@deividAlfa Does your logic analyzer show the start, stop, and repeated start conditions of i2c?

When writing the bridge I had checksum problems but finally found out that it is due to the way the PEC is calculated. Normally, when there's a read request, first there's the write operation for the address to be read and then there's the read operation for the values. When there's repeated start between the two operations, SMBus specifies that the checksum is calculated over the _whole_ stream, that is, over both the write and the read operations. Initially, my code accidentally placed normal stop and start conditions which interrupted the stream and thus the BQZ checksum was calculated only over the read operation, while `comm_sbs_bqctrl.py` expects one over both operations. The code has been fixed but just in case can you confirm that there's repeated start between the two operations?

deividAlfa, 2025-07-14:
I'll check in the afternoon!

deividAlfa, 2025-07-15:
Bingo! It was the stupid Arduino library.
So, I made my own in Low Level libs. Working as it should now! No more checksum errors.

LEaign the code, should easily adapt to most Cortex MCUs (STM32, etc).
I used this with a PY32F002A + [IOsetting libs](https://github.com/IOsetting/py32f0-template/)

Thanks for pointing it out @Cleric-K, I would have never noticed that!

### main.c
    #include "main.h"
    
    cmd_state_t state;
    ErrorStatus s;
    uint8_t i2c_bf[BFSZ], addr, i, len, write, stop;
    uint16_t out;
    volatile uint8_t rx_bf[BFSZ], in, count, bad_data;
    volatile uint32_t tick;
    
    uint32_t getTick(void){
      return tick;
    }
    
    void init_i2c(void){
      LL_APB1_GRP1_DisableClock(LL_APB1_GRP1_PERIPH_I2C1);
      LL_APB1_GRP1_ForceReset(LL_APB1_GRP1_PERIPH_I2C1);
      LL_APB1_GRP1_ReleaseReset(LL_APB1_GRP1_PERIPH_I2C1);
      LL_APB1_GRP1_EnableClock(LL_APB1_GRP1_PERIPH_I2C1);
      LL_I2C_Init(I2C1, &(LL_I2C_InitTypeDef){LL_I2C_MAX_SPEED_STANDARD, LL_I2C_DUTYCYCLE_16_9, 0, LL_I2C_NACK});
    }
    
    void sysInit(void){
      LL_RCC_HSI_Enable();
      LL_RCC_HSI_SetCalibFreq(LL_RCC_HSICALIBRATION_24MHz);
      while(LL_RCC_HSI_IsReady() != 1);
    
      LL_RCC_SetAHBPrescaler(LL_RCC_SYSCLK_DIV_1);
    
      LL_RCC_SetSysClkSource(LL_RCC_SYS_CLKSOURCE_HSISYS);
      while(LL_RCC_GetSysClkSource() != LL_RCC_SYS_CLKSOURCE_STATUS_HSISYS);
    
      LL_FLASH_SetLatency(LL_FLASH_LATENCY_0);
    
      LL_RCC_SetAPB1Prescaler(LL_RCC_APB1_DIV_1);
    
      LL_SetSystemCoreClock(24000000);
    
      SysTick_Config(240000);
      NVIC_SetPriority(SysTick_IRQn, 0);
    
      LL_IOP_GRP1_EnableClock(LL_IOP_GRP1_PERIPH_GPIOA | LL_IOP_GRP1_PERIPH_GPIOB | LL_IOP_GRP1_PERIPH_GPIOF);
    
      setLED(RED, RESET);
      LL_GPIO_Init(RED_Port, &(LL_GPIO_InitTypeDef){ RED_Pin, LL_GPIO_MODE_OUTPUT, LL_GPIO_SPEED_FREQ_VERY_HIGH, LL_GPIO_OUTPUT_PUSHPULL, LL_GPIO_PULL_NO });
    
      LL_GPIO_Init(RX_Port, &(LL_GPIO_InitTypeDef){ RX_Pin, LL_GPIO_MODE_ALTERNATE, LL_GPIO_SPEED_FREQ_VERY_HIGH, LL_GPIO_OUTPUT_PUSHPULL, LL_GPIO_PULL_NO, RX_AF });
      LL_GPIO_Init(TX_Port, &(LL_GPIO_InitTypeDef){ TX_Pin, LL_GPIO_MODE_ALTERNATE, LL_GPIO_SPEED_FREQ_VERY_HIGH, LL_GPIO_OUTPUT_PUSHPULL, LL_GPIO_PULL_NO, TX_AF });
      LL_APB1_GRP2_EnableClock(LL_APB1_GRP2_PERIPH_USART1);
      LL_USART_Init(USART1, &(LL_USART_InitTypeDef){ 921600, 8, 1, LL_USART_PARITY_NONE, LL_USART_DIRECTION_TX_RX, LL_USART_HWCONTROL_NONE, LL_USART_OVERSAMPLING_16 });
      LL_USART_ClearFlag_TC(USART1);    
      NVIC_SetPriority(USART1_IRQn,0);
      NVIC_EnableIRQ(USART1_IRQn);
      LL_USART_EnableIT_RXNE(USART1);
      LL_USART_Enable(USART1);
    
      LL_GPIO_Init(SCL_Port, &(LL_GPIO_InitTypeDef){ SCL_Pin, LL_GPIO_MODE_ALTERNATE, LL_GPIO_SPEED_FREQ_VERY_HIGH, LL_GPIO_OUTPUT_OPENDRAIN, LL_GPIO_PULL_UP, SCL_AF });
      LL_GPIO_Init(SDA_Port, &(LL_GPIO_InitTypeDef){ SDA_Pin, LL_GPIO_MODE_ALTERNATE, LL_GPIO_SPEED_FREQ_VERY_HIGH, LL_GPIO_OUTPUT_OPENDRAIN, LL_GPIO_PULL_UP, SDA_AF });
      init_i2c();
    }
    
    void put_uart(uint8_t d){
[... 219 lines cut ...]
      {
      }
    }
    
    /**
      * @brief This function handles System service call via SWI instruction.
      */
    void SVC_Handler(void)
    {
    }
    
    /**
      * @brief This function handles Pendable request for system service.
      */
    void PendSV_Handler(void){
    }
    
    void SysTick_Handler(void){
      tick++;
    }
    
    void USART1_IRQHandler(void)
    {
      UsartISR();
    }

Cleric-K, 2025-07-16:
> What? The attached traces clearly show the decoded i2c transaction?

The attached files contain only the `.pvs` (no `.sr`) which is only the session metadata. But anyway, the screenshots were sufficient.

> Bingo! It was the stupid Arduino library. So, I made my own in Low Level libs. Working as it should now! No more checksum errors.
> 
Wow :) Quite a work you've put into this. Great job! Wouldn't it have been easier to get a regular Arduino? :D

> Thanks for pointing it out [@Cleric-K](https://github.com/Cleric-K), I would have never noticed that!

Yeah, I also got that almost accidentally. I don't remember the exact scenario, I think I have been testing in parallel with CP2112 and narrowed it down to a single request which worked in one case not in the other. Analyzing the logic traces revealed that the stop conditions were the only difference. Then GPT hinted about the way PEC is calculated with or without repeated start.

So in the end did you manage to revive the battery?

In my case I'm trying to resurrect Mavic Pro batteries. Everything worked fine, I manually charged them, the unsealing worked fine, clearing PF too. Now they charge normally, however, the Mavic does not recognize them as genuine batteries and refuses to arm (there's a way to allow that but smart functions like calculating the needed battery for a return to home no longer work). As far as I could research it, it seems that this is no longer a problem in the BQZ chip but in the MSP430 mc, which seems to have triggered its own flags. But if anyone knows how this could be a matter of BQZ settings, I'm all ears.

deividAlfa, 2025-07-16:
Infixed the 3rd cell showing 0mV, the battery pad was splitted so you can solder the cell safely and join them together.

Now all the cells are reading 3800mV +-5mV.

But, unsealing doesn't work at all.
Changing the sha1 to random numbers hice the same result, so probably the key is wrong.

Can the key be tested wherher it's correct or not?

Cleric-K, 2025-07-16:
> Can the key be tested whether it's correct or not?

I don't know. Maybe you can try `FullAccess` instead of `Unseal`.

deividAlfa, 2025-07-21:
The CP21112 arrived today, gave a try to DJI Battery Killer 0.5.
Unseal didn't work either, but FullAccess did.

deividAlfa, 2025-07-21:
It turns out there was a "chemical fuse" after all.
I replaced it by 2x 1A PTC resettable fuses.

[image]
[image]

Cleric-K, 2025-07-22:
Impressive work! Do you plan on tracing the exact reasons why the two tools give different results? It is interesting for me too but I won't have a chance for delving into this for at least few more weeks.