o-gs/dji-firmware-tools #443: Requesting to add support for DJI Mini 3 Pro

persibal, 2025-08-22:
Hello everyone. 

Is there a support for DJI Mini 3 Pro to fix gimble cal issue? 

Thank you

jorgefelico, 2025-08-22:
+1 I'm also in need of the DJI Mini 3 Pro.

BuilderA7, 2025-08-24:
Does anyone happen to know if and when support for the mavic 4 pro will be
added? If there are redundancies with other mavic series I can use for M400
please advise

Xenocatalyst, 2025-08-28:
Yeah I'm looking to repair my mini 3 pro as well.

alives2000, 2025-10-23:
I can do the gimbal calibration for this model. Contact me via WhatsApp https://wa.me/+46720169505 or Telegram https://t.me/djianderson

> Does anyone happen to know if and when support for the mavic 4 pro will be
> added? If there are redundancies with other mavic series I can use for M400
> please advise
> […](#)